BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #76: Volumio installation failes

kerwingroot, 2017-12-12:
Hi, i'm running into some problems installing the Bluetooth feature for Volumio (running on RB 3b) .
Installing trough SSH seems to be working fine, until it breaks down (and reboots) ending the session with this:

```
===> Executing: su volumio -c ./bt_pa_config.sh
mkdir: cannot create directory '/home/pi/pyScripts': No such file or directory
===> Executing: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
sudo: no tty present and no askpass program specified
Exiting script due to error from: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Exiting script due to error from: su volumio -c ./bt_pa_config.sh
--------------------------------------------
./install.sh: line 291: [: !=: unary operator expected
Ending at @ Tue Dec 12 12:48:59 UTC 2017
 
```

If anybody could elaborate on this, that would be great!

BaReinhard, 2017-12-12:
What happens is that you don’t have a home folder for pi. You can either edit the script to edit pi to your user name or run the command `mkdir /home/pi`

kerwingroot, 2017-12-12:
Thanks for the quick reply!
Made the /home/pi folder, ran the config script. Took some time installing but still no show.
Is there any way to verify the deamon is running? (Sorry, pretty new to the whole RB thing)

BaReinhard, 2017-12-12:
its all good, its my fault I didnt test for Volumio explicitly and pulled it to the master branch.

you can try
the following suggested here:

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/71

Its the 10th comment down made by adkp, he ended up having to compile bluez, he listed all the commands he used to fix the issue.

BaReinhard, 2018-01-04:
I created a new branch for volumio users, heres how to try it out. Needs testing, but it has worked on my Pi 3.

```
# enable ssh through volumio.local/dev
# ssh to volumio

git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
git checkout volumio-fix
sudo ./install.sh

# Choose Bluetooth Option, #4

# Wait for it to complete, and enjoy
```

biva, 2018-01-04:
Great, it almost works :) I can connect to volumio through Bluetooth, but I don't hear anything :( Any idea?

BaReinhard, 2018-01-04:
That’s the one part I haven’t had some time to play around with much.

When playing music on the phone that’s connected, run `sudo pactl list sinks` what’s the output for it?

biva, 2018-01-08:
Output of `sudo pactl list sinks`
<pre><code>Sink #0
        State: RUNNING
        Name: alsa_output.platform-soc_audio.analog-stereo
        Description: bcm2835 ALSA Analog Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 1
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 56210 /  86% / -4.00 dB
        Monitor Source: alsa_output.platform-soc_audio.analog-stereo.monitor
        Latency: 100866 usec, configured 100000 usec
        Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "bcm2835 ALSA"
                alsa.id = "bcm2835 ALSA"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "bcm2835 ALSA"
                alsa.long_card_name = "bcm2835 ALSA"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "platform-soc:audio"
                sysfs.path = "/devices/platform/soc/soc:audio/sound/card0"
                device.string = "hw:0"
                device.buffering.buffer_size = "17640"
                device.buffering.fragment_size = "1760"
                device.access_mode = "mmap"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "bcm2835 ALSA Analog Stereo"
                alsa.mixer_name = "Broadcom Mixer"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output: Analog Output (priority: 9900)
        Active Port: analog-output
        Formats:
                pcm

Sink #1
        State: SUSPENDED
        Name: alsa_output.platform-soc_sound.analog-stereo
        Description: BossDAC Analog Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 2
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.platform-soc_sound.analog-stereo.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "Boss DAC HiFi [Master] pcm512x-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "1"
                alsa.card_name = "BossDAC"
                alsa.long_card_name = "BossDAC"
                alsa.driver_name = "snd_soc_allo_boss_dac"
                device.bus_path = "platform-soc:sound"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
                device.string = "hw:1"
                device.buffering.buffer_size = "17640"
                device.buffering.fragment_size = "1764"
                device.access_mode = "mmap"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "BossDAC Analog Stereo"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output: Analog Output (priority: 9900)
        Active Port: analog-output
        Formats:
                pcm
</code></pre>

BaReinhard, 2018-01-08:
While the device is connected and playing audio via bluetooth run the following command:

```
sudo pactl move-sink-input `sudo pactl list sink-inputs | grep "Sink Input" | sed "s/Sink Input #//"` 1
```
If that works I’ll provide you with an updated instruction to finish it off.

biva, 2018-01-08:
Thank you @BaReinhard , the output is `You have to specify a sink input index and a sink`

BaReinhard, 2018-01-08:
To me that shows that there isn’t a sink-input.

What’s the output of `sudo pactl list sink-inputs`?

biva, 2018-01-08:
<pre><code>
Sink Input #2
        Driver: module-loopback.c
        Owner Module: 16
        Client: n/a
        Sink: 0
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65537 / 100% / 0.00 dB,   front-right: 65537 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 62298843 usec
        Sink Latency: 0 usec
        Resample method: speex-fixed-1
        Properties:
                media.role = "abstract"
                module-stream-restore.id = "sink-input-by-media-role:abstract"
                media.name = "Loopback from SPECTRE"
                media.icon_name = "audio-card-bluetooth"

Sink Input #5
        Driver: module-loopback.c
        Owner Module: 17
        Client: n/a
        Sink: 0
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65537 / 100% / 0.00 dB,   front-right: 65537 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 62375441 usec
        Sink Latency: 0 usec
        Resample method: speex-fixed-1
        Properties:
                media.role = "abstract"
                module-stream-restore.id = "sink-input-by-media-role:abstract"
                media.name = "Loopback from SPECTRE"
                media.icon_name = "audio-card-bluetooth"

Sink Input #8
        Driver: module-loopback.c
        Owner Module: 18
        Client: n/a
        Sink: 0
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65537 / 100% / 0.00 dB,   front-right: 65537 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 62244852 usec
        Sink Latency: 0 usec
        Resample method: speex-fixed-1
        Properties:
                media.role = "abstract"
                module-stream-restore.id = "sink-input-by-media-role:abstract"
                media.name = "Loopback from SPECTRE"
                media.icon_name = "audio-card-bluetooth"
</code></pre>

biva, 2018-01-08:
Output of `sudo pactl list sources`:
<pre><code>Source #0
        State: IDLE
        Name: alsa_output.platform-soc_audio.analog-stereo.monitor
        Description: Monitor of bcm2835 ALSA Analog Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 1
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor of Sink: alsa_output.platform-soc_audio.analog-stereo
        Latency: 0 usec, configured 100000 usec
        Flags: DECIBEL_VOLUME LATENCY
        Properties:
                device.description = "Monitor of bcm2835 ALSA Analog Stereo"
                device.class = "monitor"
                alsa.card = "0"
                alsa.card_name = "bcm2835 ALSA"
                alsa.long_card_name = "bcm2835 ALSA"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "platform-soc:audio"
                sysfs.path = "/devices/platform/soc/soc:audio/sound/card0"
                device.string = "0"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Formats:
                pcm

Source #1
        State: SUSPENDED
        Name: alsa_output.platform-soc_sound.analog-stereo.monitor
        Description: Monitor of BossDAC Analog Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 2
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor of Sink: alsa_output.platform-soc_sound.analog-stereo
        Latency: 0 usec, configured 0 usec
        Flags: DECIBEL_VOLUME LATENCY
        Properties:
                device.description = "Monitor of BossDAC Analog Stereo"
                device.class = "monitor"
                alsa.card = "1"
                alsa.card_name = "BossDAC"
                alsa.long_card_name = "BossDAC"
                alsa.driver_name = "snd_soc_allo_boss_dac"
                device.bus_path = "platform-soc:sound"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
                device.string = "1"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Formats:
                pcm

Source #2
        State: RUNNING
        Name: bluez_source.00_28_F8_1D_69_4B
        Description: SPECTRE
        Driver: module-bluez5-device.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 15
        Mute: no
        Volume: front-left: 58982 /  90% / -2.75 dB,   front-right: 58982 /  90% / -2.75 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor of Sink: n/a
        Latency: 76974084 usec, configured 135294 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                bluetooth.protocol = "a2dp_source"
                device.description = "SPECTRE"
                device.string = "00:28:F8:1D:69:4B"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                bluez.path = "/org/bluez/hci0/dev_00_28_F8_1D_69_4B"
                bluez.class = "0x2a010c"
                bluez.alias = "SPECTRE"
                device.icon_name = "audio-card-bluetooth"
        Ports:
                unknown-input: Bluetooth Input (priority: 0, available)
        Active Port: unknown-input
        Formats:
                pcm
</pre></code>

biva, 2018-01-08:
Output of `sudo pactl list cards`:
<pre><code>

Card #0
        Name: alsa_card.platform-soc_audio
        Driver: module-alsa-card.c
        Owner Module: 1
        Properties:
                alsa.card = "0"
                alsa.card_name = "bcm2835 ALSA"
                alsa.long_card_name = "bcm2835 ALSA"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "platform-soc:audio"
                sysfs.path = "/devices/platform/soc/soc:audio/sound/card0"
                device.string = "0"
                device.description = "bcm2835 ALSA"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Profiles:
                output:analog-mono: Analog Mono Output (sinks: 1, sources: 0, priority: 200, available: yes)
                output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6000, available: yes)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: output:analog-stereo
        Ports:
                analog-output: Analog Output (priority: 9900, latency offset: 0 usec)
                        Part of profile(s): output:analog-mono, output:analog-stereo

Card #1
        Name: alsa_card.platform-soc_sound
        Driver: module-alsa-card.c
        Owner Module: 2
        Properties:
                alsa.card = "1"
                alsa.card_name = "BossDAC"
                alsa.long_card_name = "BossDAC"
                alsa.driver_name = "snd_soc_allo_boss_dac"
                device.bus_path = "platform-soc:sound"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
                device.string = "1"
                device.description = "BossDAC"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Profiles:
                output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6000, available: yes)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: output:analog-stereo
        Ports:
                analog-output: Analog Output (priority: 9900, latency offset: 0 usec)
                        Part of profile(s): output:analog-stereo

Card #2
        Name: bluez_card.00_28_F8_1D_69_4B
        Driver: module-bluez5-device.c
        Owner Module: 15
        Properties:
                device.description = "SPECTRE"
                device.string = "00:28:F8:1D:69:4B"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                bluez.path = "/org/bluez/hci0/dev_00_28_F8_1D_69_4B"
                bluez.class = "0x2a010c"
                bluez.alias = "SPECTRE"
                device.icon_name = "audio-card-bluetooth"
        Profiles:
                a2dp_source: High Fidelity Capture (A2DP Source) (sinks: 0, sources: 1, priority: 10, available: yes)
                headset_audio_gateway: Headset Audio Gateway (HSP/HFP) (sinks: 1, sources: 1, priority: 20, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_source
        Ports:
                unknown-output: Bluetooth Output (priority: 0, latency offset: 0 usec, not available)
                        Part of profile(s): headset_audio_gateway
                unknown-input: Bluetooth Input (priority: 0, latency offset: 0 usec, available)
                        Part of profile(s): a2dp_source, headset_audio_gateway

</pre></code>

BaReinhard, 2018-01-08:
It is odd you should not have multiple sink inputs. What device are you streaming from?

Also I would try the following:

`sudo pactl list sink-inputs` look at all the sink input numbers then run the following commands

`sudo pactl move-sink-input THENUMBER 1`
Replace THENUMBER with the sink input numbers, do this for each sink input, listen for a couple seconds in between each command. You should find a input that plays sound.

biva, 2018-01-08:
Thank you for your support. I'm streaming from SPECTRE.

Great, number 5 is working :)

BaReinhard, 2018-01-08:
In your file /usr/local/bin/bluez-udev there is a line audio_sink=0 change that to 1. And the streaming should be automatic

biva, 2018-01-08:
Yup, it's working, thank you so much.

I also have the impression that after a few hours without connection, or when I restart my computer, the bluetooth is unavailable. It comes back after I restart my RPI3. Is there a command to restart / initialize the bluetooth?

BaReinhard, 2018-01-08:
You can try 

`sudo systemctl restart bluetooth-agent` 

Although it should stay on indefinitely.

BaReinhard, 2018-01-08:
Although another issue is that you should disconnect your bt device when done instead of just shutting it down. That likely will prevent that issue.

closed 2018-01-08 by BaReinhard

biva, 2018-01-08:
Thank you so much, can't wait that you can make a plugin (I saw your other issue, don't hesitate if you need me to test anything). It would even be better if @volumio could integrate it in the volumio image.

BaReinhard, 2018-01-08:
I’ll be doing some development today for the plugin. I already made another branch specifically for it, so it’ll just take formatting things into the format of the plugin. Seems pretty straight forward from what I’ve read on it. I’ll definitely reach out for testing and make an issue on how to use it.