BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #77: Internet connection for install 1 (car)

ddu942, 2017-12-12:
Hello, good job for project .Sorry my very bad English.
I would like to know if it was possible to add an internet connection to the installation number 1 (car)
via wifi usb stick or other (tethering iphnone usb)

Regards Damien

BaReinhard, 2017-12-12:
Thank you, and no worries about your english.

I'm not 100% what you are wanting.

What is the goal of having the tethering or wifi connection? What do you want to do that isn't available with the current setup?

ddu942, 2017-12-13:
my project is to make a car pc
so I use option 1
but I can not put more internet on the raspberry pi3
I'm looking for a solution
silk via a usb wifi silk usb

BaReinhard, 2017-12-18:
when you set up a raspberry pi as an access point you don't have any internet access. Adding this functionality however is probably out side the scope of this project as I don't see many having a use for it. I believe there are walkthroughs for what you are looking for, you just have to google something along the lines of 'raspberry pi as Access point with internet'

closed 2018-01-11 by BaReinhard