BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #82: Install fails: typo and other issues in sound_card_install.sh

Volkov84, 2017-12-18:
Hi!

Today I've tried to install the Receiver, but failed @ sound_card_install.sh.
I found a typo in the file, (line 6: if [ -z "$SoundCard ] ) but after correcting it, it still fails. "ext missing".
Sorry, I don't have more logs, I've immediately downloaded the previous stable version, that's is working fine :)

Fresh image: 2017-07-05-raspbian-jessie-lite.img, on rpi3.

BaReinhard, 2017-12-18:
#83 I made some changes to the soundcard_install. You can try the following:

```
# cd to the ssrpari directory
git checkout soundcard-fix
sudo ./soundcard_install.sh
```

Should run correctly now. It had many incorrect conditional statements, incorrect shebang, and likely causing issues because of improper logging as well. It should be fixed now. Let me know what you find.

Volkov84, 2017-12-18:
I've made a fresh new install to avoid any conflicts.
The soundcard_install.sh looks ok, install continues on that point.
Now the install crash at:

SSRPARI [Mon 18 Dec 17:44:05 UTC 2017] Running ./kodi_install.sh...
SSRPARI [Mon 18 Dec 17:44:05 UTC 2017] Installing kodi...
SSRPARI [Mon 18 Dec 17:48:09 UTC 2017] Executing cd /home/kodi
Fatal error encountered: 'cd /home/kodi' failed
Fatal error encountered: './kodi_install.sh' failed
pi@raspberrypi:~/Super-Simple-Raspberry-Pi-Audio-Receiver-Install $

BaReinhard, 2017-12-18:
I see where the issue came from and removed it. Additionally made some changes to kodi config.sh and prevented any issues from occuring there as well (as best as I can see).

Kodi scripts should now work.

closed 2017-12-18 by BaReinhard

Volkov84, 2017-12-18:
SSRPARI [Mon 18 Dec 19:02:11 UTC 2017] Running ./kodi_install.sh...
SSRPARI [Mon 18 Dec 19:02:11 UTC 2017] Installing kodi...
SSRPARI [Mon 18 Dec 19:04:13 UTC 2017] Running ./kodi_config.sh...
SSRPARI [Mon 18 Dec 19:04:13 UTC 2017] Executing usermod -a -G input kodi
Fatal error encountered: 'usermod -a -G input kodi' failed
Fatal error encountered: './kodi_config.sh' failed

Still having some trouble.
(The user 'kodi' does not exist.)

BaReinhard, 2017-12-18:
I changed the file to look like:

```
#!/bin/bash

if [ -z "$exc" ]
then
    source functions.sh
    source dependencies.sh
fi
exc adduser kodi pulse-access
exc usermod -a -G input kodi
exc usermod -a -G lp kodi
exc usermod -a -G pulse-access,audio root

exc sed -i "s/ENABLED=0/ENABLED=1/" /etc/default/kodi 
# Add gpu_mem=256 to config.txt for use with Kodi
echo gpu_mem=256 >> /boot/config.txt
# Faster Boot
exc cp boot/cmdline.txt /boot/cmdline.txt
exc touch ~/.hushlogin

```
What you can do is manually add the user kodi

`sudo adduser kodi`
then re run kodi script as root. This file may need some love to fix. I'll look into it for sure.

BaReinhard, 2017-12-18:
I did a little testing, and what might actually need to happen is just removed any user modifications from the file so it would look like:

```
#!/bin/bash

if [ -z "$exc" ]
then
    source functions.sh
    source dependencies.sh
fi

exc sed -i "s/ENABLED=0/ENABLED=1/" /etc/default/kodi 
# Add gpu_mem=256 to config.txt for use with Kodi
echo gpu_mem=256 >> /boot/config.txt
# Faster Boot
exc cp boot/cmdline.txt /boot/cmdline.txt
exc touch ~/.hushlogin
```

or simply create a kodi user before hand.

Volkov84, 2017-12-18:
After Kodi, the next stop:

SSRPARI SSRPARI [Mon 18 Dec 20:16:55 UTC 2017] Running ./gmrender_install.sh...
./gmrender_install.sh: line 16: exc: command not found
./gmrender_install.sh: line 17: exc: command not found
./gmrender_install.sh: line 19: exc: command not found
./gmrender_install.sh: line 20: exc: command not found
./gmrender_install.sh: line 22: exc: command not found
./gmrender_install.sh: line 23: exc: command not found

BaReinhard, 2017-12-18:
Thank you, another bug,

The first line

`if [ -z “exc” ]` should be `if [ -z “$exc” ]`

Sorry for all the bugs here. I appreciate you’re thorough inspection.

Volkov84, 2017-12-19:
I thank you the great and fast support! I just need a working Christmax background music solution :)
Currently I have a partially working SSRPAR (only Airplay working, but fine for me now) but after Christmas I can do a little more testing.
Just a little add-on:
- if the install fails for the first time, it will never run again. It fails. For example the "exc mkdir" because the directory is already there. And there are a couple of more command where it stops.
- if I manually run a separate script (eg: gmrender_install.sh) it will run and do it's job, until there is a variable needed from the main script (eg: line 27: $GMediaName)

BaReinhard, 2017-12-19:
If you do have time, i created a new branch called `multi-exec` which fixes the gmedia install, but due to patch commands I will have to find a way to allow for multiple executions which will prevent from exiting due to patches not working. I have an idea, but it will take a little time to implement.

Volkov84, 2017-12-19:
Och, sorry.. Didn't want to give you extra job. I don't think the "multi-exec" function is needed, don't waste effort on that. If the install works fine, there should be no need for individual executions.

If the install fails on an "mkdir" is a real issue, it should check if the directory exists or not.

Tomorrow morning I'll try again on a fresh Raspbian image.

Should I stay with jessie or can I try stretch (2017-11-29)?

Volkov84, 2017-12-19:
Couldn't sleep...
A new try on a fresh jessie lite:

SSRPARI [Tue 19 Dec 21:14:18 UTC 2017] Running su pi -c ./snapcast_install.sh...
Would you like to install SnapCast as a Server(s), Client(c), or both (b)?: (s/c/b) c
SSRPARI [Tue 19 Dec 21:18:45 UTC 2017] Executing git clone https://github.com/badaix/snapcast.git
SSRPARI [Tue 19 Dec 21:18:58 UTC 2017] Executing cd /home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/snapcast/externals
SSRPARI [Tue 19 Dec 21:18:58 UTC 2017] Executing git submodule update --init --recursive
SSRPARI [Tue 19 Dec 21:21:40 UTC 2017] Executing sudo apt-get install build-essential -y
SSRPARI [Tue 19 Dec 21:21:42 UTC 2017] Executing sudo apt-get install libasound2-dev libvorbisidec-dev libvorbis-dev libflac-dev alsa-utils libavahi-client-dev avahi-daemon -y
Fatal error encountered: 'sudo apt-get install libasound2-dev libvorbisidec-dev libvorbis-dev libflac-dev alsa-utils libavahi-client-dev avahi-daemon -y' failed
Fatal error encountered: 'su' failed

BaReinhard, 2017-12-19:
@Volkov84 I agree, but it allows for users to better debug issues in the event there is a bug/error. In addition it gives me a better way of trying multiple times without running into issues. I have found a solution for the mkdir problems and is in the latest `multi-exec` branch. There are some remnants in snapcast that will need to be gutted.

Also don't feel bad for giving me extra work, its what I like doing and it's something that has been requested before but never had the time to implement or the knowledge to do so. Now I have both and can spent some time on it.

BaReinhard, 2017-12-19:
Alright the `multi-exec` branch has ( I believe, full back ups and full uninstall capabilities). It will create a new folder for backing up all the original files that will be changed.

Another feature is restoring these files
```
source functions.sh
restore_originals
```
It will restore all original files that were backed up.

Of course to gain this functionality you will need to use the install on the newer branch

```
git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
git checkout multi-exec
sudo ./install.sh
```

Additionally, this branch should fix many of the errors you have been experiencing. (But alas, I am sure there are more. Its a wonder how this script ever worked) :smile: