BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #84: Volume a bit low

jacobdu215, 2017-12-18:
So I am currently testing this with my car and it seems the volume output (AUX) seems very low. So I tested it on a speaker I use with my computer and it also seems to be very quiet. Is there a reason for this? How can I raise the volume?

BaReinhard, 2017-12-18:
Is this with a sound card or built ?

BaReinhard, 2017-12-20:
take a look here: https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/bluez-udev#L26

You may need to change the 0 to the soundcard number found via

```
sudo pactl list sinks
```

It also has been a while since i've played around with amixer, but that may be the issue here: 
https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/bluez-udev#L24

`numid=3` may need to be changed to a different number

To get a better understanding for changing volume via amixer, check out the page here: https://linux.die.net/man/1/amixer

I believe the amixer lines could probably be removed from bluez-udev since we are using pulseaudio for playback, but I haven't tested this.

dawithers, 2017-12-23:
The problem is with the built-in audio

BaReinhard, 2017-12-23:
If increasing the volume with 

`sudo alsamixer`

Doesn’t fix it, then it’s likely the built in audio. You can also try increasing the source input volume but that should be set to 100% anyways.

dawithers, 2017-12-23:
Yeah it's at 100. Bummer. Thanks!

BaReinhard, 2017-12-23:
Sound cards are cheap and worth the investment. If you want a really cheap one a usb sound card from sabrent would be a great addition. $6 on Amazon

BaReinhard, 2017-12-23:
For another ~$10+ a hifiberry or iq audio is worth the extra money.

BaReinhard, 2018-01-11:
You might also try looking at setting the pulseaudio sink and source volumes.

closed 2018-01-11 by BaReinhard