BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #86: multi-exe branch needs testing!!

BaReinhard, 2017-12-20:
## ***HELP WANTED***
For those who would like to help contribute, I have created a new branch `multi-exec` it was created to help serve the purpose of debugging during install and to create the ability to uninstall the project if no longer desired. There are a number of new scripts and tools to help along this process. Many of which need testing to ensure they work across the board and to catch any syntax errors.

## Install

```
git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
git checkout multi-exec
sudo ./install.sh
```

## Uninstall *(after Install)*

```
cd super-simple-raspberry-pi-audio-receiver-install
source functions.sh
source dependencies.sh
full_uninstall
```

## Restore Originals *(not needed if Uninstall was run)*

```
cd super-simple-raspberry-pi-audio-receiver-install
source functions.sh
source dependencies.sh
restore_originals
```

kirilldobr, 2017-12-21:
Jessie Lite
Crashes after running `full_uninstall` and then `sudo ./install.sh`<img width="1280" alt="screen shot 2017-12-21 at 5 04 54 pm" src="https://user-images.githubusercontent.com/8530571/34258904-224cf2e4-e671-11e7-9d47-df9c4ede0f81.png">

BaReinhard, 2017-12-21:
the error in /etc/pulsebackup has been fixed, via this line: https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/multi-exec/bt_pa_config.sh#L165

Kodi issue has been fixed with the following: https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/multi-exec/kodi_config.sh#L8

I am unsure if Kodi creating a user is done on reboot or whether that feature was removed in recent updates. For now it will be removed, it was originally done to allow for audio playback in kodi via pulseaudio. They may have chosen to use alsa as the sound option instead. This will be a feature I will continue to test and keep an eye on. 

For now comment out the lines I showed in kodi_config.sh and make the addition in the bt_pa_config.sh. Then run ./install.sh again, it should make its way further down the installation.

BaReinhard, 2017-12-22:
Spent a lot of time testing this today.

I have also included Snapcast Support, with server and Client with matching configuration. Some enduser knowledge is needed (i.e. for clients needing to know the Snapserver IP Address), Additionally, depending on the DAC being used, some configuration to `/etc/default/snapclient` will be needed for -s #, you can find it via `snapclient --list`. Just setup a multiroom audio setup with 2 commands.

korte84, 2017-12-22:
Hi @BaReinhard 
i've tested this package and it encounter a fatal error when trying to run bt config 
```
...
SSRPARI [Fri 22 Dec 11:53:28 UTC 2017] Executing cd pulseaudio
SSRPARI [Fri 22 Dec 11:53:28 UTC 2017] Executing sudo ./bootstrap.sh
Fatal error encountered: 'sudo ./bootstrap.sh' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed
```

BaReinhard, 2017-12-22:
@korte84 what os are you running?

korte84, 2017-12-26:
Jessie (but I have to test it again)

vudev, 2017-12-31:
I had sample problem
Fatal error encountered: 'sudo ./bootstrap.sh' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed
raspbian stretch

BaReinhard, 2017-12-31:
Will not work with stretch. This is still for Jessie only.

mdrons, 2018-01-02:
Are there good directions on how to put Jessie on a brand new Pi 3?  The current version of NOOBS only offers Stretch.

nilayp, 2018-01-02:
This is a good primer on installing OS images on an SD card: https://www.raspberrypi.org/documentation/installation/installing-images/

The last version of Jessie can be downloaded here: http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/2017-07-05-raspbian-jessie.zip

-- Nilay

BaReinhard, 2018-01-03:
thanks @nilayp , when I get some time I’ll either add a how to or link somewhere in the readme.  @mdrons thanks for asking this question. It never occurred to me to put a guide on this.

closed 2018-01-04 by BaReinhard