BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #87: No sound output with HiFiBerry Amp+

MKSounds, 2017-12-20:
Hello,
I was really glad I found exactly the solution I was looking for to handle the most common audio tasks on the raspberry with one piece of software.
The installation process on my Raspberry PI Zero W with Raspbian Jessie Lite ran through without any problems (although it took more than 2 hours).
I choose the HiFiBerry Amp+ during the installation and it is installed correctly (as aplay says).
The problem is, I don't get sound output, either using bluetooth or via AirPlay. Both connecting processes seem to work on my IPhone.

I already tried to set the volume of the master output and of the soundcard in the alsa mixer, but that didn't fix the issue.
I also tried speaker test with: 
`speaker-test -Dhw:0,0 -c2 -twav`
That worked (at a low volume and with bad audio quality).

Is there any other method to check the audio throughput from bluetooth or AirPlay ?

Kind regards,
Markus.

BaReinhard, 2017-12-21:
there are a couple of things you can try. I am assuming things work fine on the alsa side of things as aplay says, but bluetooth is using pulseaudio so we must check there first.
it'll be easiest to check when you are connect via bluetooth and streaming audio.

```
sudo pactl list sinks
# should display sinks, if dtparam=audio=on is commented out in the /boot/config.txt you should only see your hifi card. look at the Sink #

sudo pactl list sink-inputs
# should display 1 sink input with a specific sink input #,

sudo pactl move-sink-input sink-input# sink#
```
Audio should start playing, if the sink# is different than 0 or this line https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/bluez-udev#L2

change the file /usr/local/bin/bluez-udev and change it to the sink#

MKSounds, 2017-12-21:
I did as you mentioned (see below for the results). The sink number is 0, which is normal when using a DAC or AMP+.
I don't really understand what you mean with "Audio should start playing, if the sink# is different than 0".
In the file "/usr/local/bin/bluez-udev" the variable audio_sink is set to 0, which should be correct.
```
pi@raspberrypi:~ $ sudo pactl list sinks
Sink #0
        State: IDLE
        Name: alsa_output.0.analog-stereo
        Description: snd_rpi_hifiberry_amp Analog Stereo
        Driver: module-alsa-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 1
        Mute: no
        Volume: front-left: 17027 /  26% / -35.12 dB,   front-right: 17027 /  26% / -35.12 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.0.analog-stereo.monitor
        Latency: 51100 usec, configured 66666 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "HifiBerry AMP HiFi tas5713-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "snd_rpi_hifiberry_amp"
                alsa.long_card_name = "snd_rpi_hifiberry_amp"
                alsa.driver_name = "snd_soc_hifiberry_amp"
                device.string = "hw:0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "snd_rpi_hifiberry_amp Analog Stereo"
                device.icon_name = "audio-card-analog"
        Formats:
                pcm
pi@raspberrypi:~ $ sudo pactl list sink-inputs
Sink Input #0
        Driver: module-loopback.c
        Owner Module: 15
        Client: n/a
        Sink: 0
        Sample Specification: s16le 2ch 43943Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 17027 /  26% / -35.12 dB,   front-right: 17027 /  26% / -35.12 dB
                balance 0.00
        Buffer Latency: 373574 usec
        Sink Latency: 63870 usec
        Resample method: speex-fixed-1
        Properties:
                media.role = "abstract"
                module-stream-restore.id = "sink-input-by-media-role:abstract"
                media.name = "Loopback from iPhone"
                media.icon_name = "audio-card-bluetooth"

```

BaReinhard, 2017-12-21:
you're correct. I was assuming the card showed up as a different sink number. Its odd to me that audio isn't playing then. You have all the correct things showing up. In the case that something in bluez-udev isn't working, have you tried the following command? `sudo pactl move-sink-input 0 0`

Assuming the input hasn't changed number.

Also what is the output of `sudo amixer`?

MKSounds, 2017-12-21:
The move-sink-input didn't cause any changes.
I suspect there could be a volume problem. 
Output of alsa mixer is:
```
pi@raspberrypi:~ $ sudo amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 11869 [18%] [on]
  Front Right: Playback 11869 [18%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 58982 [90%] [on]
  Front Right: Capture 58982 [90%] [on]

```

MKSounds, 2017-12-21:
The amixer seems not to be affected by this command:
```
pi@raspberrypi:~ $ sudo amixer sset 'Master',0 volume 100%
pi@raspberrypi:~ $ sudo amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 22191 [34%] [on]
  Front Right: Playback 22191 [34%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 58982 [90%] [on]
  Front Right: Capture 58982 [90%] [on]

```

BaReinhard, 2017-12-21:
you're right, it I guess I tried too many things at once without checking. 

It appears that `sudo alsamixer` will correctly change volume levels for both alsa and pulseaudio

MKSounds, 2017-12-23:
Yes with `sudo alsamixer` it's possible to set the volume.
But that didn't change the situation. I think there must be a problem directing the audio data to the sound output (because both Bluetooth and AirPlay are not working).

Therefore I checked in /etc/shairport-sync.conf if the hardware is set correctly:
` output_device = "hw:0,0"; ` is okay.

MKSounds, 2017-12-23:
Is there any possibility to get a debug report of the whole Audio Receiver while it's running ?

BaReinhard, 2017-12-23:
No, but I’m looking into adding such a feature for this reason. 

Have you taken a look at your `/etc/asound.conf` file?

MKSounds, 2017-12-23:
I tried the audio receiver on a Pi 2B with a HifiBerry DAC+ and things are running fine.
I will test with the DAC+ on the Pi Zero W next, to see if the problem is caused by the HifBerry Amp.

BaReinhard, 2017-12-24:
What was the contents of /etc/asound.conf?

MKSounds, 2017-12-24:
On the working system on the 2B it is:
```
pi@raspberrypi:~ $ sudo nano /etc/asound.conf
  GNU nano 2.2.6                                         File: /etc/asound.conf

pcm.pulse {
    type pulse
    card 0
}
ctl.pulse {
    type pulse
    card 0
}

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}
```
After it worked with the 2B and the DAC+ I tried if it works with the AMP+. So I changed the device tree overlay to hifiberry-amp and enabled i2c and i2s (in boot/config.txt). I thought this should work, because it should not be relevant if soundcard 0 is the amp or the dac.
But that didn't work neither.
Is it possible to change the soundoutput/card after the installation or has the soundcard to be integrated during installation ?

The contents of asound.conf on the Zero W with AMP+ is (which is not working):
```
pcm.pulse {
    type pulse
    card 0
}
ctl.pulse {
    type pulse
    card 0
}

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}
```
which is exactly the same.

BaReinhard, 2017-12-24:
You shouldn’t need to enable the i2c and i2s in addition to the dtoverlay. The dtoverlay should be enough

MKSounds, 2017-12-27:
Yeah you are right, i2c ans i2s don't have to be activated.
But that shouldn't affect the problem with the hifiberry amp+.
I will try if the DAC+ works in combination with the Pi Zero W.

MKSounds, 2017-12-30:
I tried it on the Pi Zero W with the DAC+ and it worked. If I change the dtoverlay to the Amp+, the Amp is initialised correctly via i2c (you can her the noise increase when it's unmuted), but there's no sound output (hardware is tested working).
I already tried a new clean install (always with Option 6). Do I have to install the UPnP Renderer to get stuff running ?
Do you have any suggestions what I could check ?

BaReinhard, 2017-12-30:
can you tell me the output of the following command:

```
cat $SSRPARI_PATH/install_choices
```

MKSounds, 2017-12-31:
That file is not available on both installations:
```
pi@raspberrypi:~ $ cat $SSRPARI_PATH/install_choices
cat: /install_choices: No such file or directory
```

BaReinhard, 2017-12-31:
Ok sorry, it looks like there’s a typo in the install script it should be the above, but until it gets fixed use

```
cat $SSPARI_PATH/install_choices
```

Sorry 😐

MKSounds, 2017-12-31:
I was already wondering why it's SSPARI :)
```
pi@raspberrypi:~ $ cat $SSPARI_PATH/install_choices
[Sat 23 Dec 21:57:54 UTC 2017] Running cat...
Bluetooth = y
AirPlay = y
AP = n
Kodi = n
Lirc = y
SoundCardInstall = y
GMedia = n
```
As I mentioned, I just need Bluetooth audio and AirPlay in combination with a soundcard.

BaReinhard, 2017-12-31:
I would try a new install without lirc. I am assuming this may be interfering with the amp+

MKSounds, 2018-01-01:
I tried a new install without lirc. That haven't changed the problem.
There must be an issue between the audio receivers (Bluetooth and AirPlay) and the sound output (alsa ?), because both don't work and the init of the hifiberry amp+ is correct.

```
pi@raspberrypi:~ $ cat $SSPARI_PATH/install_choices
[Mon  1 Jan 19:03:57 UTC 2018] Running cat...
Bluetooth = y
AirPlay = y
AP = n
Kodi = n
Lirc = n
SoundCardInstall = y
GMedia = n
Snapcast = n
```

indritix, 2018-01-05:
I can confirm that by changing the Sink # from 0 to 1 in "/usr/local/bin/bluez-udev" I was able to listen when connected via BT.
1 is the sink # of my hifiberry DAC

BaReinhard, 2018-01-05:
@MKSounds what is the output of `sudo pactl list sinks`?

MKSounds, 2018-01-06:
```
pi@raspberrypi:~ $ sudo pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.0.analog-stereo
        Description: snd_rpi_hifiberry_amp Analog Stereo
        Driver: module-alsa-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 1
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.0.analog-stereo.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "HifiBerry AMP HiFi tas5713-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "snd_rpi_hifiberry_amp"
                alsa.long_card_name = "snd_rpi_hifiberry_amp"
                alsa.driver_name = "snd_soc_hifiberry_amp"
                device.string = "hw:0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "snd_rpi_hifiberry_amp Analog Stereo"
                device.icon_name = "audio-card-analog"
        Formats:
                pcm
```
```
pi@raspberrypi:~ $ sudo pactl move-sink-input `sudo pactl list sink-inputs | grep "Sink Input" | sed "s/Sink Input #//"` 0
You have to specify a sink input index and a sink
```

BaReinhard, 2018-01-06:
@mksounds , when you ran the command we’re you streaming audio to the device?

It looks like no device was connected or at least playing any music.

BaReinhard, 2018-01-06:
So here’s how it should work,

Connected your phone or music device to the Pi via Bluetooth. Once it’s connected start playing music.

After that you should be able to see a sink input via
`sudo pactl list sink-inputs` . If nothing shows there, there might be a problem with something else. Possibly a bad solder on the board, I have my hifiberry amp+ working as of yesterday with the new install. I’ll test again this weekend to see what happens.

MKSounds, 2018-01-07:
The hardware of the amp+ is working correctly. I tested it with Volumio.

```
pi@raspberrypi:~ $ sudo pactl list sink-inputs
Sink Input #0
        Driver: module-loopback.c
        Owner Module: 15
        Client: n/a
        Sink: 0
        Sample Specification: s16le 2ch 44034Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 18580 /  28% / -32.85 dB,   front-right: 18580 /  28% / -32.85 dB
                balance 0.00
        Buffer Latency: 214538 usec
        Sink Latency: 48787 usec
        Resample method: speex-fixed-1
        Properties:
                media.role = "abstract"
                module-stream-restore.id = "sink-input-by-media-role:abstract"
                media.name = "Loopback from iPhone"
                media.icon_name = "audio-card-bluetooth"
```
That should be okay. I also tested if it's a volume issue. I set the volume to 92% via alsamixer, but that wasn't the solution.
I'm wondering why it's just me having those difficulties. Especially because with the dac+ everything is fine.

BaReinhard, 2018-01-07:
Once you can see that there is a sink input I would suggest running this command:

```
sudo pactl move-sink-input `sudo pactl list sink-inputs | grep "Sink Input" | sed "s/Sink Input #//"` 0
```

I don't know why, but everything should be working correctly. 

Aside from the `/etc/asound.conf` you should remove the card line from pulse so it should look like the following:

```
pcm.pulse {
    type pulse
}
ctl.pulse {
    type pulse
}

```

MKSounds, 2018-01-08:
I did the steps as you told me, but still no sound output.
I will check the hardware again, but I don't think it's the reason of the problem.

BaReinhard, 2018-01-08:
What would help a lot is when you see that you have a sink listed under `sudo pactl list sink-inputs` is also show me the output of `sudo pactl list sinks` As well.

MKSounds, 2018-01-08:
```
pi@raspberrypi:~ $ sudo pactl list sink-inputs
Sink Input #0
        Driver: module-loopback.c
        Owner Module: 15
        Client: n/a
        Sink: 0
        Sample Specification: s16le 2ch 43922Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 25284 /  39% / -24.82 dB,   front-right: 25284 /  39% / -24.82 dB
                balance 0.00
        Buffer Latency: 130435 usec
        Sink Latency: 54681 usec
        Resample method: speex-fixed-1
        Properties:
                media.role = "abstract"
                module-stream-restore.id = "sink-input-by-media-role:abstract"
                media.name = "Loopback from iPhone"
                media.icon_name = "audio-card-bluetooth"
pi@raspberrypi:~ $ sudo pactl list sinks
Sink #0
        State: RUNNING
        Name: alsa_output.0.analog-stereo
        Description: snd_rpi_hifiberry_amp Analog Stereo
        Driver: module-alsa-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 1
        Mute: no
        Volume: front-left: 25284 /  39% / -24.82 dB,   front-right: 25284 /  39% / -24.82 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.0.analog-stereo.monitor
        Latency: 60930 usec, configured 66666 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "HifiBerry AMP HiFi tas5713-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "snd_rpi_hifiberry_amp"
                alsa.long_card_name = "snd_rpi_hifiberry_amp"
                alsa.driver_name = "snd_soc_hifiberry_amp"
                device.string = "hw:0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "snd_rpi_hifiberry_amp Analog Stereo"
                device.icon_name = "audio-card-analog"
        Formats:
                pcm
```
That's the output while playing audio from an Iphone via bluetooth.

BaReinhard, 2018-01-08:
This is extremely odd. Can you try increasing your volume on your iPhone . Everything on the os side seems like it should be playing audio. It may be an issue with the dac, but I wouldn’t know where to start with it. I’ll test today on my hifiberry amp and see if I get similar issues.

BaReinhard, 2018-01-08:
However, I would still try opening a support thread with hifiberry for your device. It seems that your issue isn’t all that uncommon.

MKSounds, 2018-01-08:
> Can you try increasing your volume on your iPhone

Yes I already tried that. The volume adjustment on the phone is working.

BaReinhard, 2018-01-08:
I also have a couple questions/suggestions. The amp+ isn’t originally made for a pi zero, so are you using an external 12v to power the amp+? Also check the soldering on all pins of th pi zero, it is important they are good solder joints. Lastly, did you solder all of the pins on the pi zero or just a few?

MKSounds, 2018-01-08:
So far I'm using a laboratory power supply @ ~15V to power the amp. 
I just soldered the pins for i2c and i2s on the Pi Zero. The wiring was made with jumpers, for testing.
That is a good suggestion to check the solder joints. I haven't thought about that yet.

BaReinhard, 2018-01-08:
They suggest that you solder pins to the ground and 5v leads as well. So you should be able to run the pi and amp from only the amp plug.

MKSounds, 2018-03-06:
I resoldered the pins with a standard soldering iron and it didn't affect the problem. I checked everything and couldn't find another reason.
So I checked the soldered points under the microscope. I found a cold junction. I desoldered the pin and cleaned it and soldered it with a soldering iron for smd components.
It's working perfect now.
Thank's a lot for your support!

MKSounds, 2018-03-11:
@BaReinhard  the issue can be closed. Thank you for your help.