BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #88: AirPlay speed

kirilldobr, 2017-12-21:
I see you use shairport-sync as part of your project.
But there was a problem with it - when pausing/playing audio lagged behind. If you was watching YouTube, for example, you experienced an audio lag.
Does your project work similarly or did you solve the syncing issue?

BaReinhard, 2017-12-21:
Unfortunately, this issue is present in this project as well. Until they are able to find a solution for streaming video & audio, (like an appletv) this issue is likely to be present.

I am assuming that you are watching the video on a iDevice streaming audio to the pi.

The issue stems from the ability of the device to delay video until sound is played on the device. It's doubtful apple would give us that power over our devices.

closed 2017-12-21 by BaReinhard

kirilldobr, 2017-12-21:
There is a great app called LonelyScreen.
I don't know what magic are they using, but music streaming from iDevice to PC works flawlessly.

BaReinhard, 2017-12-21:
I had never seen that before. Unfortunately, it looks like it only works on Mac and PC. Unfortunately, it is also a paid service. I don't know enough about avahi or AirPlay to even begin to solve this issue either. Ultimately, this issue is more relevant to shairport-sync's issue location.

kirilldobr, 2017-12-21:
Looks like it has Sublime-like paying model, but that's behind the question)
Anyway, I will continue testing and write back with the results.

BaReinhard, 2017-12-21:
This has gotten me curious, I'll continue looking into solutions. How long is the delay between audio and video?

kirilldobr, 2017-12-21:
Just now installed shairport on my Mac and everything is working great. When i pause a video on iPhone and then start it again there is 2.25 second delay in audio and then it catches up and syncs perfectly, as explained here:
https://github.com/mikebrady/shairport-sync#synchronisation-latency-stuffing

Now I'll try to install it to raspberry again and report back. Maybe I had an old shairport version or something.

kirilldobr, 2017-12-22:
Yep, I was right!
I was installing shairport from some 7 year old repository. Installing the last version removed all my issues.

BaReinhard, 2017-12-22:
Awesome! Glad the updated version works for you!

kirilldobr, 2017-12-22:
Do you how to stream audio from windows pc to raspberry? It may be over a network or with extra hardware, just anything.

BaReinhard, 2017-12-22:
Either airfoil for windows or tuneblade