BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #90: Kodi error?

acidhax, 2017-12-24:
SSRPARI [Sat Dec 23 22:01:15 UTC 2017] Running ./kodi_install.sh...
SSRPARI [Sat Dec 23 22:01:15 UTC 2017] Checking kodi...
dpkg-query: no packages found matching kodi
SSRPARI [Sat Dec 23 22:01:15 UTC 2017] Installing kodi...
SSRPARI [Sat Dec 23 22:05:09 UTC 2017] Running ./kodi_config.sh...
SSRPARI [Sat Dec 23 22:05:09 UTC 2017] /etc/default/kodi does not exist
SSRPARI [Sat Dec 23 22:05:09 UTC 2017] Executing sed -i s/ENABLED=0/ENABLED=1/ /etc/default/kodi
Fatal error encountered: 'sed -i s/ENABLED=0/ENABLED=1/ /etc/default/kodi' failed
Fatal error encountered: ./kodi_config.sh failed

BaReinhard, 2017-12-24:
Interesting it looks like the kodi apt package has completely changed. I’ll have to look at what happened. Thanks for opening the issue.

BaReinhard, 2017-12-24:
Can you run kodi from the cli or does it open automatically on a reboot?

acidhax, 2017-12-24:
https://gist.github.com/Cyberek/33af1b92c071791a71aa8bccf87b8a3a

Makes it work. I can run kodi from command line though.

BaReinhard, 2017-12-24:
Thanks I’ll look into that as well. Since it’ll require more steps I’m going to look into installing the latest stable version of kodi as well and see what makes more sense.

acidhax, 2017-12-24:
Curious, do I really need to run Kodi just to have bluetooth audio for my car?

BaReinhard, 2017-12-24:
No, kodi is just used as a ui. Used with a remote or controller makes navigation of stored music on the pi easier to navigate. To be honest I’m thinking of removing kodi from the built in installs. If you simply want a headless install and just want to use an aux in cable without video. I would suggest doing a custom install with just Bluetooth. The added packages with kodi and other built ins will slow down your boot times, if all you’ll be using is Bluetooth just install that.

You can run the following and start over,

```
cd $SSPARI_PATH
source functions.sh
source dependencies.sh
full_uninstall

# Then run the install again.
sudo ./install.sh
```

closed 2018-01-11 by BaReinhard