BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #91: apt-get: show output during installation process

ifs-net, 2017-12-25:
updating a raspberry system might take a long time. Maybe you want to add an option to show the output of apt-get during your install.sh
Otherwise it's like a "black box", you cannot see any progress of the installation routine.

BaReinhard, 2017-12-25:
You bring up a good point. I haven’t needed it on any jessie install but doesn’t mean it isn’t needed. I’ll look into this.

BaReinhard, 2018-01-05:
A current work around for this is in the `dependencies.sh` change the install command and remove the -qq except for the -y and go to the `functions.sh` and in the function apt_install, remove ’&> /dev/null’ after INSTALL_COMMAND, this will show all the dpkg output.

closed 2018-01-05 by BaReinhard