BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #92: install.sh: Installation of package 'libjson0-dev' failed (raspbian stretch incompatibility)

ifs-net, 2017-12-25:
```
Which Sound Card are you using? (0/1/2/3/4/5/6/7/8/9/10/11) : 4
SSRPARI [Mo 25. Dez 10:08:38 CET 2017] Updating via Apt-Get
SSRPARI [Mo 25. Dez 10:08:44 CET 2017] Upgrading via Apt-Get
SSRPARI [Mo 25. Dez 10:08:46 CET 2017] Running ./bt_pa_install.sh...
SSRPARI [Mo 25. Dez 10:08:46 CET 2017] Checking python-dbus...
SSRPARI SSRPARI [Mo 25. Dez 10:08:46 CET 2017] Dependency python-dbus already me                 t...
[Mo 25. Dez 10:08:46 CET 2017] Checking pulseaudio...
SSRPARI [Mo 25. Dez 10:08:46 CET 2017] Dependency pulseaudio already met...
SSRPARI [Mo 25. Dez 10:08:46 CET 2017] Checking libtool...
SSRPARI [Mo 25. Dez 10:08:46 CET 2017] Installing libtool...
SSRPARI [Mo 25. Dez 10:08:56 CET 2017] Checking intltool...
dpkg-query: Kein Paket gefunden, das auf intltool passt
SSRPARI [Mo 25. Dez 10:08:56 CET 2017] Installing intltool...
SSRPARI [Mo 25. Dez 10:09:26 CET 2017] Checking libsndfile-dev...
dpkg-query: Kein Paket gefunden, das auf libsndfile-dev passt
SSRPARI [Mo 25. Dez 10:09:26 CET 2017] Installing libsndfile-dev...
SSRPARI [Mo 25. Dez 10:09:36 CET 2017] Checking libcap-dev...
dpkg-query: Kein Paket gefunden, das auf libcap-dev passt
SSRPARI [Mo 25. Dez 10:09:37 CET 2017] Installing libcap-dev...
SSRPARI [Mo 25. Dez 10:09:44 CET 2017] Checking libjson0-dev...
dpkg-query: Kein Paket gefunden, das auf libjson0-dev passt
SSRPARI [Mo 25. Dez 10:09:45 CET 2017] Installing libjson0-dev...
Fatal error encountered: Installation of package 'libjson0-dev' failed
Fatal error encountered: ./bt_pa_install.sh failed
```

libjson0-dev is deprecated in raspbian stretch.
sudo apt-get install libjson-c-dev will install the correct frameworks.

=> dependencies.sh should be updated (change libjson0-dev to libjson-c-dev)

BaReinhard, 2017-12-25:
Thanks for the update on dependencies, unfortunately I don’t have a current plan on updating for raspbian stretch. Still waiting to see how bluealsa ends up before implementing an update.

vudev, 2017-12-31:
I had sample problem

BaReinhard, 2018-01-05:
Will reopen, when thoughts of implementing on stretch.

closed 2018-01-05 by BaReinhard