BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #93: Weird sound on raspberry pi zero w (bluetooth)

DG4ever, 2017-12-27:
The installation worked without any problems on the zero w using "raspbian full".
Unfortionatly I am getting a bad sound when streaming via bluetooth, it sounds like an echo or so.
I tried disabling wifi but that made no difference. 
I noticed that the first 1-3 seconds after connecting sounds good but then the echo starts.
Might the zero w to weak for this application? Or is there a buffer that I could increase?

BaReinhard, 2017-12-27:
This doesn't sound like the bluetooth/wifi issue. Can you display the out of the following commands:

```
cat $SSPARI_PATH/install_choices
```

nilayp, 2017-12-27:
By any chance- did you have to run the installer more than once? In troubleshooting something a few weeks ago, I accidentally ran the installer twice and had the exact same symptom- the Bluetooth had an echo. (Airplay worked fine.)

I never debugged it to figure out why... I reflashed my SD card with a fresh Rasbian Jessie image and ran the installer once... And the problem went away.

BaReinhard, 2017-12-28:
@nilayp Multiple running of install script should now have no averse effects with the newest pull request. It restores original files and restarts the installation. This is a recent change, maybe a week or 2 at most.

Likely, what I assume the issue is here (without seeing the install_choices file) is that either multiple instances of a snapclient are running, or it is an issue with the Raspbian Full installation, the install is meant for Jessie Lite, but with the aforementioned revision with the latest pull request I will likely be able to adapt something for Jessie with Pixel in the coming weeks. Possibly next week, since my last day of work is tomorrow 😀  until the 8th of jan when I start my new job.

DG4ever, 2017-12-28:
`[Wed 27 Dec 12:06:14 UTC 2017] Running cat...
Bluetooth = y
AirPlay = n
AP = n
Kodi = n
Lirc = n
SoundCardInstall = n
GMedia = n`

Btw I am using usb headphones.

BaReinhard, 2017-12-28:
interesting, did you install Snapcast as well? I had forgotten to add the variable to `install_choices`.

I see no other reason why this should be happening aside from it being a full version of raspbian. 

I haven't ever come into contact with this similar issue.

You might want to try changing the resampling in pulseaudio

```
/etc/pulse/daemon.conf

...
# resample-method defaults to  speex-float-1 on most architectures,
# speex-fixed-1 on ARM
resample-method = speex-float-1 # use speex-fixed-1 instead and comment out ffmpeg
;resample-method = ffmpeg #prepend a ; to this line
enable-remixing = no
enable-lfe-remixing = no
...

I believe this may fix issues for the smaller boards as RPI Zero and RPI Zero W. Possibly the earlier generation boards as well.

DG4ever, 2017-12-29:
Changing the resampling didn't help, so I just flashed Raspbian lite and installed everything again,
Now the sound is great, no more echo. Thanks for your help!

I have some other question to a different topic. 
I need to mix the input from a microphone to the output (microphone is card 1). 
I tried it with:
`alsaloop -C hw:1,0 -P plug:dmix -c 1  -t 60000 -b`
or
`alsaloop -C hw:1,0 -c 1 -t 60000`

Unfortionatly I am getting "device or resource busy" when bluetooth is playing.
The weird thing is that `sudo speaker-test -c2 -twav` is working while playing over bluetooth.
Do you now how I can mix the mic input to the output?

Do you know how I can force that my usb headphones are always the main device to play from?
Whe I reboot with the headphones plugged in they become card 0. Unfortionatly when I plug them in after boot, they become card 2 and the music won't play through them.

BaReinhard, 2017-12-29:
For adding a mic, you should take a look at mixing sound input and output in alsa. That isn’t something that’s likely going to be supported so it’s likely a one off you’ll have to include.

As far as setting default indexes is easy enough. What cards do you have currently available in `aplay-l`?

totti2, 2017-12-31:
I had a flanger effect some SSRPARIs ago.
How does the output of `sudo pactl list sink-inputs` and `sudo pactl list sinks` look like?

When connecting Bluetooth sources, more than one sink-input were created automagically and routed to only one sink. One could even see different sample-rates in the verbose sink-info. Maybe, I ran the installer twice, can't tell anymore.

DG4ever, 2017-12-31:
`aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: iWear [iWear], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

This is the output. The sound should be played via the "iWear" Device. Unfortionatly the device is not always on card2. Sometimes it becomes card0.. 
I know that I can set the default card to play from in `/usr/share/alsa/alsa.conf`
But this is done by the card number and not the name.

BaReinhard, 2017-12-31:
Do you need the other two devices connected? You can remove the on board sound in the config.txt , but even if you keep them all you can still set the cards to be indexed specifically on boot.

This is an example for setting the onboard sound always as index 0 on boot

`echo "options snd_bcm2835 index=0" | sudo tee -a /etc/modprobe.d/alsa-base.conf`
 You would need to do the same for other device names. I haven’t done it but it should be do able with a little googling and time investment.

BaReinhard, 2018-01-08:
My suggestion here would be to do a fresh installation, choose the same options, but this time include soundcard and choose usb soundcard. This will setup the sound output correctly. If there are still issues, write back and let me know.

L0rd0fB0red0m, 2018-04-07:
Not sure where this should be posted, but I experience severe lagging when playing music via BT.
Every second or two there's a short interruption. 
I'm running the new RPi 3 B+, stretch (not lite), I made a clean installation on a freshly flashed system, there weren't amy errors.

mgsolid, 2020-03-17:
I met a similar issue today on my Raspberry4. After some troubleshooting I found an interesting thing and resolved the issue. Here is what I guess:
When the device (iPhone) connecting to Raspberry's bluetooth, at the first a few seconds Raspberry will decide the connection type by the signal quality. If I took the iPhone a little far away from Raspberry and connect it with bluetooth, Raspberry will be connected in a "low quality" mode, and then the sound of music would be wired all the time. But I connected iPhone very near to the Raspberry, then a "high quality" mode, and the music sounds very well.