BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #94: sudo patch 'main.conf' failed

david1234vo, 2017-12-28:
I am fairly new to Raspberry Pi's and I am trying to run your script on a model B with raspbian jessie lite but when i choose the home installation i gt these errors:

SSRPARI [Thu 28 Dec 15:30:37 UTC 2017] Executing sudo patch /etc/bluetooth/main.conf
Fatal error encountered: 'sudo patch /etc/bluetooth/main.conf' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed

BaReinhard, 2017-12-28:
can you paste the output of:

```
cat $SSRPARI_BACKUP_PATH/main.conf
```

david1234vo, 2017-12-28:
pi@raspberrypi:~ $ cat Super-Simple-Raspberry-Pi-Audio-Receiver-Install/main.con                                                                             f
cat: Super-Simple-Raspberry-Pi-Audio-Receiver-Install/main.conf: No such file or                                                                              directory

BaReinhard, 2017-12-28:
should be 

```
cat Super-Simple-Raspberry-Pi-Audio-Receiver-Install/backup_files/main.conf
```

david1234vo, 2017-12-28:
`
pi@raspberrypi:~ $ cat Super-Simple-Raspberry-Pi-Audio-Receiver-Install/backup_files/main.conf
[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
#Name = %h-%d
Name =

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=

`

david1234vo, 2017-12-28:
pi@raspberrypi:~ $ cat Super-Simple-Raspberry-Pi-Audio-Receiver-Install/backup_files/main.conf
[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
#Name = %h-%d
Name =

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=

david1234vo, 2017-12-28:
```
pi@raspberrypi:~ $ cat Super-Simple-Raspberry-Pi-Audio-Receiver-Install/backup_files/main.conf
[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
#Name = %h-%d
Name =

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=
```

mdrons, 2017-12-30:
Mine failed only because I ran the install multiple times.  It fails on this line:
  # 0 = disable timer, i.e. stay discoverable forever
! #DiscoverableTimeout = 0

because it is already commented out.  

I am running on Stretch, so not sure if that has something to do with it.

BaReinhard, 2017-12-30:
That is likely the cause, this is not usable without user configuration. Several packages have changed. There won't likely be an update for Stretch in a few months at the earliest.

kevipapo1, 2017-12-31:
For those who need a dirty workaround, you can manually make the required changes to `/etc/bluetooth/main.conf` and comment out the lines in `bt_pa_config.sh` that apply the patch to that file.

silelmot, 2018-01-02:
i have the problem, there seems not to be any main.conf. i have (after executing the install.sh) a main.conf.orig, but this one is empty

BaReinhard, 2018-01-03:
Here is a basic copy of one:

```
[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
#Name = %h-%d

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=
```

However, I am concerned as to why you don’t have one. This makes me think you’re using a different os or version of raspbian (which may be incompatible with this repo)

silelmot, 2018-01-03:
Bluetooth wasnt installed and the script didnt install it.
but after having the main.cfg the script stopped at this part and i changed the main.cfg manually

BaReinhard, 2018-01-04:
What is the output of 

```
cat /etc/os-release
```

silelmot, 2018-01-05:
sorry, already removed the os, because it doesnt worked as expected. it was the newest volumio2-release. just installed it on my new raspberry pi 3 b

BaReinhard, 2018-01-08:
Alright, I have a stretch version that is in the testing phase at the moment, go [here](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/102) and follow the instructions. It will need to be on a new installation of Raspbian Stretch, (it has been tested on the LITE version, but i suspect it will work for Stretch w/ Pixel). Please leave feedback in issue #102 .

closed 2018-01-08 by BaReinhard