BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #99: Does not install on Volumio 2.348 on RPi2

HGBAXXTER, 2018-01-05:
Thanks for this great piece of work. I have been using it on a Raspbian distribution and it worked just fine. 

However, to also use other functions I wanted to migrate to Volumio 2. Unfortunately, the install stops with the following command line output:

SSRPARI [Fri Jan  5 19:10:35 UTC 2018] Updating via Apt-Get
SSRPARI [Fri Jan  5 19:11:12 UTC 2018] Upgrading via Apt-Get
SSRPARI [Fri Jan  5 19:11:12 UTC 2018] Running ./bt_pa_install.sh...
SSRPARI [Fri Jan  5 19:11:12 UTC 2018] Checking pulseaudio-module-bluetooth...
dpkg-query: no packages found matching pulseaudio-module-bluetooth
SSRPARI [Fri Jan  5 19:11:12 UTC 2018] Installing pulseaudio-module-bluetooth...
Fatal error encountered: Installation of package 'pulseaudio-module-bluetooth' failed
Fatal error encountered: ./bt_pa_install.sh failed

I used option 4 for the install. The system is Volumio 2.348 on RPi2.
Thanks for any support!

BaReinhard, 2018-01-05:
Thanks for the report, that is odd, what does it say when you manually install pulseaudio-module-bluetooth?

```
sudo apt-get install pulseaudio-module-bluetooth
```

HGBAXXTER, 2018-01-05:
OK, thanks, I figured it out in parallel. For some reason, dpkg was corrupted on my system, which caused the script to stop. After running `sudo dpkg --configure -a` and manually installing `sudo apt-get install pulseaudio-module-bluetooth` the script seems to run just fine. Thanks for the good work and support!

BaReinhard, 2018-01-05:
glad its working. I believe there can be issues with the install command, so I made a commit to fix that issue. Shouldn't have an issue from now on.

HGBAXXTER, 2018-01-05:
Strange, I just repeated it on a fresh Volumio image. Running the script the first time, it stops (or takes reeeallly long) at "SSRPARI [Fri Jan  5 21:45:41 UTC 2018] Upgrading via Apt-Get".

After rebooting, it again stops with "dpkg-query: no packages found matching pulseaudio-module-bluetooth".

Initially I thought it was a single time effect, but it does not seem to be the case.

BaReinhard, 2018-01-05:
thats output is because it checks for a pulseaudio-module-bluetooth package is previously installed, if it is it won't install it. It shouldn't effect anything, does it still error out afterwards? Upgrading Apt-get will take a long time as it needs to update all its packages from Raspbian Jessie, which is a pretty long list.

HGBAXXTER, 2018-01-06:
Do you have an estimation of how long apt-get upgrade will take? I started again from scratch and it is now stuck on upgrade for almost an hour now. 

Is it possible to modify the script so that it will display the output of apt-get upgrade to see the exact progress?

Thanks

BaReinhard, 2018-01-06:
Yes, and I’m looking at adding a verbose output option.

Just remove the `&> /dev/null` from this line:
https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/install.sh#L267

BaReinhard, 2018-01-08:
the above comment seemed to fix the issue for other Volumio users.

I have made a new branch `stretch-fix` which includes fixes for this logging as well and should fix the issue. It will still work for volumio users as well. Take a look at issue #102 for install instructions.

closed 2018-01-08 by BaReinhard

HGBAXXTER, 2018-01-08:
Thanks, I will give it a try soon.

BaReinhard, 2018-01-08:
I’ll be working on a volumio plug-in this afternoon hopefully can have it working later this evening. Once it’s done I’ll be sure to add an issue for volumio users to check out

HGBAXXTER, 2018-01-08:
Even better, if you need somebody to test it, let me know. I am regularly using a raspberry with DAC for bluetooth playback and at the moment I am looking for the best way of doing so. Volumio with Plug-In would be the preferred solution.

HGBAXXTER, 2018-01-21:
I finally had some more time test and think I understand why the installation stops: during apt-get upgrade it asks when removing `&> /dev/null`:

Setting up dhcpcd5 (6.7.1-1+rpi5) ...

Configuration file '/etc/dhcpcd.conf'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** dhcpcd.conf (Y/I/N/O/D/Z) [default=N] ?

I am running on an RPi2 with USB WLAN and USB BT

HGBAXXTER, 2018-01-21:
The stretch-fix fails with `Fatal error encountered: Installation of package 'liborc-0.4-dev' failed`

Sadly, I am not able to get a working solution for Volumio :-(

BaReinhard, 2018-01-21:
The volumio-fix branch should work. You may need to update and upgrade manually before running the install but it did work for me at one point.