BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #100: Issue setting up Bluetooth Only Installation

Utmostmoney, 2018-01-07:
I am coming across this issue during the installation process 

`SSRPARI [Sun  7 Jan 05:41:30 UTC 2018] Checking libcap-dev...
SSRPARI [Sun  7 Jan 05:41:30 UTC 2018] Dependency libcap-dev already met...
SSRPARI [Sun  7 Jan 05:41:30 UTC 2018] Checking libjson0-dev...
dpkg-query: no packages found matching libjson0-dev
SSRPARI [Sun  7 Jan 05:41:30 UTC 2018] Installing libjson0-dev...
Fatal error encountered: Installation of package 'libjson0-dev' failed
Fatal error encountered: ./bt_pa_install.sh failed
`

I am pretty new to this, and I am not quite sure how to go about fixing this. Any ideas?

BaReinhard, 2018-01-07:
This has been reported before. Are you running raspbian stretch or installed raspbian from the NOOBS image? These operating systems are not supported you need to install the latest raspbian Jessie image to get this repo working. There is a link to the image in the README.md file.

BaReinhard, 2018-01-07:
Tomorrow I’ll be creating a markdown file to follow for installing for both Mac and windows users if you don’t already know how to do so. You can either wait or search google on how to do so. The link for the image is here: http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/

Utmostmoney, 2018-01-07:
I am using the Raspbian lite download from the website, and did `sudo apt-get update` and `sudo apt-get upgrade` before installing git to clone the repo. 

I guess I will wait for tomorrow to try reinstalling with the markdown you make to see if it fixes the problem.

BaReinhard, 2018-01-07:
What happens when you install libjson0-dev manually?

sudo apt-get install libjson0-dev

Utmostmoney, 2018-01-07:
It says that it is unable to locate the package

pi@raspberrypi:~/Super-Simple-Raspberry-Pi-Audio-Receiver-Install $ sudo apt-get install libjson0-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libjson0-dev

BaReinhard, 2018-01-07:
I misread your previous comment, if you're talking about the Raspberry Pi website in the download section that is Raspbian Stretch, to get a Jessie image you need to use this link:
http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/

Raspbian Stretch requires a different libjson package, if you remove that package from the dependencies.sh and insteade use `libjson-c-dev` you shouldn't get the error. I am unsure if stretch will work correctly, but I have had other reports that it does. That being said, I have no tested it

BaReinhard, 2018-01-08:
Alright, I have a stretch version that is in the testing phase at the moment, go [here](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/102) and follow the instructions. It will need to be on a new installation of Raspbian Stretch, (it has been tested on the LITE version, but i suspect it will work for Stretch w/ Pixel). Please leave feedback in issue #102 .

BaReinhard, 2018-01-08:
Any issues or feedback in regards to raspbian stretch please redirect to issue #102

closed 2018-01-08 by BaReinhard

Utmostmoney, 2018-01-08:
Well, too bad that wasn't posted a bit earlier. Lol. I already downloaded the Jessie image and got it working wonderfully. 

I will be sure to test it with the Stretch version though. As a slight side question, it is possible to get the Bluetooth to automatically connect to my phone once the pi is turned on? (I use an iPhone if that would be relevant)

BaReinhard, 2018-01-08:
Haha yeah sorry, I finally broke down and decided since NOOBS uses stretch I may as well spend some time looking into it. Found out it was far easier than I expected also since I chose to stick with pulseaudio even though stretch seems to want to use bluez-alsa instead.

Given that your phones bluetooth is on when the pi boots it should be pretty easy to achieve.

Take a look at my stackoverflow question I asked and answered myself 😨  . [Here](https://raspberrypi.stackexchange.com/questions/44520/bluetooth-autoconnect-on-reboot-car-setup-with-scripts/44539#44539) . This should be able to achieve what you want. If it doesn't let me know and I can start looking at some testing in my free time.

Utmostmoney, 2018-01-08:
@BaReinhard Also, one final question from me for the night. I was doing all this testing on my Pi3, and when I moved it to my Pi0 W the audio was not working properly, would a simple flash and resetup fix this issue? 

I'll look into the stackoverflow answer tomorrow after school. Thanks again!

BaReinhard, 2018-01-08:
You shouldn't have an issue setting up on a Pi3 and moving to a 0W, this is how I do all my setup.

It will likely depend on what you install though, what are the install options and what audio card are you using?

Utmostmoney, 2018-01-08:
All I'm using is the bluetooth only install with no audio card.

BaReinhard, 2018-01-08:
hmm, that may be related to what the output of the device is for `aplay -l` on a Zero W I am unsure of what the base sound card is and if it defaults to outputting to the hdmi output. I have only ever used a Zero W with a sound card of some sort.

4runl0g, 2018-01-15:
Iam using stetch install, getting below exceptions

SSRPARI [Mon Jan 15 18:59:02 UTC 2018] File 'main.conf' has been previously backed up
SSRPARI [Mon Jan 15 18:59:02 UTC 2018] Executing sudo patch /etc/bluetooth/main.conf
Fatal error encountered: 'sudo patch /etc/bluetooth/main.conf' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed

Kindly help

BaReinhard, 2018-01-15:
Are you running stretch with Pixel or lite?

JHoelzel, 2018-09-21:
Iam Running into the same issues on lite:
`
SSRPARI [Fri 21 Sep 11:40:18 UTC 2018] Executing sudo patch /etc/bluetooth/main.conf
Fatal error encountered: 'sudo patch /etc/bluetooth/main.conf' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed
`
am now trying to burn the card again and try from complete scratch

JHoelzel, 2018-09-21:
So i just wanted to let you know, i fresh installed lite and it worked out of the box (after upgrade + git installed) so i suppoed if this ever happens again, its pebcac and the user has modifiied the bt config beforehand.

However once i sudo reboot ed i get this when opening bluetoothctl:

`
pi@raspberrypi:~ $ bluetoothctl
[bluetooth]# nnect to bluetoothd...
`

JHoelzel, 2018-09-21:
so yeah i got the last issue solved as well!
When the devices are connected they are not automatically trusted!!!!!

I just trusted my device and am hearing blissfull music from my pi3 b+

The Error Message about the SAP profiles can be ignored, its based on contact exchange from your phone!

So after all thanks for putting this all together, its not really detailrich but that let me down a crusade of bluetoothctl, the hdaudio output, routing and bluetooth trusts.

Hearing my sound comming from my pi just made my day ;)