BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #102: **** Support for Raspbian Stretch, TESTING NEEDED!!! ****

BaReinhard, 2018-01-08:
## Testing Needed / Help Wanted

I have a working version for Raspbian Stretch Lite, unsure if it works for Stretch with Pixel (I am assuming it will, but haven't tested).

I need testers, and feedback.

## Install

```
git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
git checkout stretch-fix
sudo ./install.sh
```

mdrons, 2018-01-08:
Installed this morning on a Pi3 with Stretch Lite and everything completed successful.  I did copy a sample /etc/default/kodi file in.   It seems kodi does not include this file with install anymore.  I am a little confused on why or what a sound card provides.   Is it explained anywhere?  If I was to buy, which one is best?  Do they support different features?  Out of the 10 supported, how do I pick one?  Thanks.

BaReinhard, 2018-01-08:
Thanks for the feedback. Yeah kodi has changed, it needs some fixing along with Access Point.

Sound cards install basically will configure the output for the Bluetooth and airplay and setup an asound.conf file.

Sound cards themselves provide a higher quality/fidelity to the audio. The built in sound is ok but isn’t as rich. Most of the ones I’ve used (iqaudio, hifiberry,  and Sabrent usb all work far better than the built in, with iqaudio and hifiberry beating out sabrent) there are many more available online and they’re worth the small investment if you have decent speakers to output to.

silelmot, 2018-01-08:
i tried it with 1: INstallt he Raspberry Pi Audio Reciever Car Installation
and i got an error:
```
SSRPARI [Mo 8. Jan 19:15:07 CET 2018] Running ./kodi_install.sh...
SSRPARI [Mo 8. Jan 19:15:07 CET 2018] Checking kodi...
dpkg-query: Kein Paket gefunden, das auf kodi passt
SSRPARI [Mo 8. Jan 19:15:07 CET 2018] Installing kodi...
SSRPARI [Mo 8. Jan 19:16:19 CET 2018] Running ./kodi_config.sh...
SSRPARI [Mo 8. Jan 19:16:19 CET 2018] /etc/default/kodi does not exist
SSRPARI [Mo 8. Jan 19:16:19 CET 2018] Executing sed -i s/ENABLED=0/ENABLED=1/ /etc/default/kodi
Fatal error encountered: 'sed -i s/ENABLED=0/ENABLED=1/ /etc/default/kodi' failed
Fatal error encountered: ./kodi_config.sh failed
```

BaReinhard, 2018-01-08:
Thanks, yeah kodi has changed in the past year so that will need some fixing. I’ve been debating on whether or not to continue to support it. It seems that there are better options of installing kodi. Unsure of what version installs via apt-get.

I believe I’ll be removing kodi from any built in installs until I find a solid solution.

mdrons, 2018-01-08:
Use this file
 more /etc/default/kodi
# Set this to 1 to enable startup
ENABLED=1

# The user to run Kodi as
USER=pi

# Adjust niceness of Kodi (decrease for higher priority)
NICE=-5

E3V3A, 2018-01-09:
@mdrons Please use triple back-ticks ``` to identify code. Can also be followed by `bash` for highlight.

Where to find that kode file?

mdrons, 2018-01-09:
https://gist.github.com/Cyberek/33af1b92c071791a71aa8bccf87b8a3a

Thanks, Mike

SoAG, 2018-01-09:
Just tried to install with option 4. bluetooth only and got this:

```
SSRPARI [Tue Jan  9 21:13:11 UTC 2018] Saving /etc/pulse/system.pa...                                                                                                                                              
system.pa=/etc/pulse/                                                                                                                                                                                              
SSRPARI [Tue Jan  9 21:13:11 UTC 2018] Executing sudo patch /etc/pulse/system.pa                                                                                                                                   
SSRPARI [Tue Jan  9 21:13:11 UTC 2018] Executing cd /home/pi                                                                                                                                                       
SSRPARI [Tue Jan  9 21:13:12 UTC 2018] Executing remove_dir json-c                                                                                                                                                 
SSRPARI [Tue Jan  9 21:13:12 UTC 2018] Executing git clone https://github.com/json-c/json-c.git                                                                                                                    
SSRPARI [Tue Jan  9 21:13:18 UTC 2018] Executing cd json-c                                                                                                                                                         
SSRPARI [Tue Jan  9 21:13:18 UTC 2018] Executing sh autogen.sh                                                                                                                                                     
Fatal error encountered: 'sh autogen.sh' failed                                                                                                                                                                    
Fatal error encountered: su pi -c ./bt_pa_config.sh failed 
```

Manually cloning json-c and running `sh autogen.sh` give this error:

```
configure.ac:50: installing './compile'             
configure.ac:12: installing './config.guess'        
configure.ac:12: installing './config.sub'          
configure.ac:6: installing './install-sh'           
configure.ac:140: error: required file './ltmain.sh' not found                                           
configure.ac:6: installing './missing'              
Makefile.am: installing './depcomp'                 
parallel-tests: installing './test-driver'          
autoreconf: automake failed with exit status: 1 
```
Any ideas?

BaReinhard, 2018-01-09:
@SoAG that’s odd haven’t come across that issue before. Are you running stretch with pixel?

Utmostmoney, 2018-01-10:
After installing the Bluetooth Only Installation on the Rpi0 and got this error:

```SSRPARI [Tue  9 Jan 21:55:38 UTC 2018] Running ./bt_pa_install.sh...
SSRPARI SSRPARI [Tue  9 Jan 21:55:39 UTC 2018] Raspbian Stretch Found
[Tue  9 Jan 21:55:39 UTC 2018] Checking libjson-c-dev...
dpkg-query: no packages found matching libjson-c-dev
SSRPARI [Tue  9 Jan 21:55:39 UTC 2018] Installing libjson-c-dev...
SSRPARI [Tue  9 Jan 21:56:03 UTC 2018] Checking autopoint...
dpkg-query: no packages found matching autopoint
SSRPARI [Tue  9 Jan 21:56:03 UTC 2018] Installing autopoint...
Fatal error encountered: Installation of package 'autopoint' failed
Fatal error encountered: ./bt_pa_install.sh failed```

After what felt like ages, it finally finished the custom install using just the bluetooth with no sound card. I connected my phone only to find out that there was a static and no music.

BaReinhard, 2018-01-10:
Is this a raspbian stretch lite or w/ pixel. Also where did you get the Pi image? Noobs or direct download from raspberrypi.org?

SoAG, 2018-01-10:
@BaReinhard I've tried again on a freshly flashed raspbian stretch and now it all worked.

BaReinhard, 2018-01-10:
@SoAG This is great news. What did you have installed previously? Anything specific you can think of?

SoAG, 2018-01-10:
I was running raspbian-stretch-lite (2017-11-29), but had previously played around with pulseaudio (installed the version from apt repository) as well as mpd and some minor testing getting a bluetooth audio sink to work before discovering this repo.

Everything seems to work now (except for some stuttering audio but that seems to be down to firmware issues). One thing to I've noted though is that there is a group `pulse-access` so I had to add my mpd and pi user to it in order to get them to play audio. What's the reason for having the separate group?

Utmostmoney, 2018-01-10:
@BaReinhard It is the Lite directly from the raspberrypi.org

BaReinhard, 2018-01-10:
@utmostmoney , what is the output when you manually install auto point?

`sudo apt-get install autopoint -y` ?

vanupied, 2018-01-11:
Awesome !!! As you can imagine it work like a charm for me :)
I just had to modify the file bt_pa_config.sh in order to replace exc git clone git://github.com/erikd/libsndfile.git by exc git clone https://github.com/erikd/libsndfile.git in order to correct the following issue: 
SSRPARI [Thu 11 Jan 10:53:05 UTC 2018] Executing git clone git://github.com/erikd/libsndfile.git
Fatal error encountered: 'git clone git://github.com/erikd/libsndfile.git' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed
I use a rpi3 with latest raspbian, I choosed to install with following choice:
Which installation would you like to choose? (1/2/3/4/5/6) : 2
Do you want all the Devices to use the same name? (y/n) : y
Device name: myaudio
Do you want to use an AirPlay password? (y/n) : n
Which Sound Card are you using? (0/1/2/3/4/5/6/7/8/9/10/11) : 0
Many many thanks for your great work !
I now will look to edit the quality because I hear a breath when no music is played but that's a detail.

BaReinhard, 2018-01-11:
@vanupied thanks for the feedback, I Didn’t even realize I used git inplace of https there. I’ll look into making that change.

@soag the reason for using that group is because pulseaudio is run as a system daemon and requires that group for users to execute the pulse commands.

Utmostmoney, 2018-01-12:
@BaReinhard Sorry for the long reply, but the Pi just tells me that auto point is up to date and nothing is updated/upgraded

BaReinhard, 2018-01-12:
@Utmostmoney no worries, I guess its possible that there are some issues when redirecting output to  /dev/null.

You can try again with removing `&> /dev/null` in the functions.sh 
on this line: https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/stretch-fix/functions.sh#L53

remove that portion and run the install again.

Utmostmoney, 2018-01-12:
@BaReinhard  Still the same output as last time

```
 $ sudo apt-get install autopoint -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
autopoint is already the newest version (0.19.8.1-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

BaReinhard, 2018-01-12:
oh sorry, i meant with that change you should be able to install the full repo. @Utmostmoney

Utmostmoney, 2018-01-12:
@BaReinhard I deleted the directory and used git to get a new copy of the repo because of some bluetooth discovery error, and I made the change you said. The Bluetooth Only Install went smoothly, but the static is still there just like the custom install. It sounds like feedback from reading/writing data (keep in mind that I still don't use a audio card of any kind), and music still will not play through the 0W

BaReinhard, 2018-01-12:
Onboard sound has always been an issue compounded with the issue of the broadcom chip of Wifi and Bluetooth causing choppy playback but you should still hear music.

Can you run the following commands and show me the ouput of each, when your bluetooth device is connected and playing music:

```
sudo pactl list sink-inputs

sudo pactl list sinks

cat /var/log/syslog

cat /etc/asound.conf

# then finally try this to see if music starts to play
sudo pactl move-sink-input `sudo pactl list sink-inputs | grep "Sink Input" | sed "s/Sink Input #//"` 0

```

old-square-eyes, 2018-01-13:
Clean install - Stretch with desktop, custom install, BT, Airplay and USB soundcard. Install finished OK. Phone pairs over BT, plays but no audio is heard through Mbox Micro.

BaReinhard, 2018-01-13:
This makes sense, this was originally created for CLI/headless installs. I am unsure what package is in control of bluetooth on the Desktop.

old-square-eyes, 2018-01-13:
All good re: CLI. Any ideas why selecting USB audio doesn't default to card 1? Also why Airplay isn't working?

old-square-eyes, 2018-01-14:
And it's stopped working again. All I did was move it into another room. This is what was happening to me on Jesse. It just crashes on a whim.

BaReinhard, 2018-01-14:
@old-square-eyes
Connect your device and play audio then run the following commands:

```
sudo pactl list sink-inputs

sudo pactl list sinks

sudo pactl list sources

aplay -l

cat /var/log/syslog

cat /etc/asound.conf
```
Then post back your output this will be helpful in diagnosing the issue.

vanupied, 2018-01-14:
Not sure since 2 days, that doesn't seems to work properly anymore. When I tried yesterday to reinstall the whole thing I had a very crappy sound over bluetooth (I just can't listen it). This occurred since last two days  of commit. Every time I do my test by reinstalling everything, the OS and SSRPAR, from scratch, but since two days that doesn't work as it used to 4 days ago (my first try and post :) )

4runl0g, 2018-01-16:
Headsup on this repo. I am trying to connect my RPi to my Sony home Theatre. First I was trying with OS installed from Rasbian, it didn't work. As suggested by @SoAG tried with fresh install with option 4(Bluetooth Only), then it started showing bluetooth device in Android mobile, but no sound. But it was not showing in iPhone 7. Then I tried installing option 6 (Custom installation with below selections), it started showing in my iPhone (Music-> Select song -> AirPlay). Then I selected RPiAudio, still no sound. 

Do you want to install SnapCast? (y/n): y
Do you want AirPlay Enabled? (y/n) : y
Do you want Bluetooth A2DP Enabled? (y/n) : y
Do you want to use a Sound Card? (y/n) : n
Would you like to install SnapCast as a Server(s), Client(c), or both (b)?: (s/c/b) b
Would you like to install Librespot (Warning: No Longer Maintained, install time: long) (y/n): n
Do you want all the Devices to use the same name? (y/n) : y
Device name: RPiAudio
Do you want to use an AirPlay password? (y/n) : n

Installation was success, but i am not getting sound on any devices. I tried connection earphone to Pi audio o/p port (Blue colour jack)

Kindly help me

4runl0g, 2018-01-16:
After executing below command, it worked. 
sudo amixer -c 0 cset numid=3 1 (Got this from Issue#52) - https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/52

Still AirPlay isn't working. Anyone has any success on this. Plz update

mdrons, 2018-01-16:
Airplay works great for me.   BT is not working.   I am having authentication issues.  
Jan 15 19:36:45 music pulseaudio[443]: [pulseaudio] protocol-native.c: Denied access to client with invalid authentication data.

When I run this as user pi:
busctl tree org.bluez
Failed to introspect object / of service org.bluez: Access denied
No objects discovered.

it works if I run it as root
 busctl tree org.bluez
└─/org
  └─/org/bluez
    └─/org/bluez/hci0
      └─/org/bluez/hci0/dev_FC_65_DE_0B_EF_B2

SoAG, 2018-01-16:
@mdrons if I run the following command the user `pi` belongs to the group `pulse-access`.

```
pi@raspberrypi:~ $ getent group | grep pulse-access
pulse-access:x:115:root,mpd,pi
```

I had to add `pi` manually though by running

```
sudo adduser pi pulse-access
```

So maybe if you do that it may fix your issue.

mdrons, 2018-01-16:
I had that.   What I just did and it seems to have fixed it (more testing required).

I copied all of the lines under the root user in /etc/dbus-1/system.d/bluetooth.conf to the bluetooth group.  

I am able to run the above command and bt streaming is working again.

BaReinhard, 2018-01-17:
Hmm this seems to have happened a few times already. I’ll have to check if the user is being exported properly

fingertrouble, 2018-01-24:
SSRPARI [Wed 24 Jan 17:59:36 GMT 2018] Executing adduser pulse audio
SSRPARI [Wed 24 Jan 17:59:36 GMT 2018] Executing adduser root pulse-access
SSRPARI [Wed 24 Jan 17:59:38 GMT 2018] Executing adduser pulse lp
SSRPARI [Wed 24 Jan 17:59:38 GMT 2018] Running su butter -c ./bt_pa_config.sh...
SSRPARI [Wed 24 Jan 17:59:39 GMT 2018] Executing sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
Fatal error encountered: su butter -c ./bt_pa_config.sh failed

I'm using a RPI2 with Stretch and Pixel, all updated to latest firmware/versions etc. 

I expected issues with pulseaudio but I purged that recently. I do have shairport-sync already installed, but finding BT audio refuses to work so hence trying this. Do you have to use pulseaudio for audio-sinks then? I'd prefer to use ALSA if at all possible. BT Audio and Airplay used to work, but recently stopped working...Hence purging and reinstalling the BT stacks, and purging Pulseaudio.

BaReinhard, 2018-01-28:
@fingertrouble the repo uses pulseaudio, I haven’t found a way to automate connection with alsa, hence the usage of pulseaudio. Bluetooth playback works with alsa but is a bit more involved and will require some cli commands. I don’t recall what they are off hand though.

johans86, 2018-01-31:
Hello,
First of all I'll just say that I'm a bit of a newbie when it comes to RPI and UNIX.

Second. Great work!

**My setup**

Raspberry PI 3 model B
JustBoom Amp HAT
Raspbian Stretch with desktop

JustBoom and DuckDNS were setup before beginning this installation.

**My devices**

Compuer with Windows 10
Phone with Android 7.0

**My installation**

Which installation would you like to choose? (1/2/3/4/5/6) : 2
Do you want all the Devices to use the same name? (y/n) : y
Device name: lg5music
Do you want to use an AirPlay password? (y/n) : y
Which Sound Card are you using? (0/1/2/3/4/5/6/7/8/9/10/11) : 10

The installation went very smooth without any error messages. After rebooting the system a message said that the system had been updated.

**Testing**

I've been testing to play sound from both the compuer and phone after connecting to the RPI using bluetooth. It has been working, but the sound quality is sometimes poor and is sometime stuttering. After disconnecting the devices sound from example the browser running on the in Raspbina can not be heard (which isn't much of a problem).

I do not see any high CPU load when the stuttering occurs. Maybe a longer buffer time could help?

I have not tried it with AirPlay yet, but will report here when I do.

Tell me if I can be of any help!

Thanks for an easy to install repo. Great work! Best regards Johan.

BaReinhard, 2018-02-05:
I’m glad to see it’s working on desktop version as well. Currently, I’m swamped with several work projects so I don’t have much time to do any dev work here. As for the choppiness, is this device connected to WiFi or an Ethernet cable?

johans86, 2018-02-05:
I had the RPI connected both by ethernet cable and WiFi. After turning off the WiFi the sound quality increased a lot and the stuttering has basically disappeared. Thanks for the tip!

BaReinhard, 2018-02-07:
No problem, it’s a known issue with the raspberry pi. I’m hoping one day soon there will be a fix

justfry, 2018-02-12:
How can I automatically restart that process without disconnection.
All works fine, but after 30-40 min of using sound gets glitchy. I manually restart whole server and it again works fine. 
I've tried to` sudo pulse -k `  and then `sudo pulseaudio -D`, but it return a message like this:
`E: [pulseaudio] main.c: Failed to kill daemon: No such process`

justfry, 2018-02-12:
O, sorry, I've got it!
In `/etc/pulse/default.pa` should be commented `#load-module module-suspend-on-idle`
It prevent to stop pulse audio when there is no signal.

fpunktz, 2018-02-16:
Hello,
I am trying to install this on OSMC (which is based on debian jessie) and get the following error when trying to install the bluetooth-a2dp option:
SSRPARI [Fri Feb 16 17:26:32 CET 2018] Executing tst sudo cp /tmp/machine-info /etc
pulseaudio
SSRPARI [Fri Feb 16 17:26:32 CET 2018] Saving /etc/init.d/pulseaudio...
pulseaudio=/etc/init.d/
SSRPARI [Fri Feb 16 17:26:32 CET 2018] Executing sudo cp init.d/pulseaudio /etc/init.d/pulseaudio
SSRPARI [Fri Feb 16 17:26:32 CET 2018] Executing sudo chmod +x /etc/init.d/pulseaudio
SSRPARI [Fri Feb 16 17:26:32 CET 2018] Executing sudo update-rc.d pulseaudio defaults
Fatal error encountered: 'sudo update-rc.d pulseaudio defaults' failed
Fatal error encountered: su osmc -c ./bt_pa_config.sh failed

Any advice?

OhmniD, 2018-02-17:
Hi,

Pi and Linux newbie here so go easy! Firstly thanks for this great project, got a Pi Zero W and HifiBerry Zero last week and have been playing around with them. Installed Stretch via NOOBS and got Shairport working nicely on my own, but wanted to get bluetooth working too. Researched it but decided there must be an easier way, tried Max2Play and didn't like it, then stumbled across this - seemed like the perfect solution. 

Blanked my SD card back to Stretch Lite and put your tester branch on, tried setting up as both Home and Custom with just Airport and bluetooth - Airport works flawlessly, my BT devices connect but no sound coming out. Did some fiddling but got nowhere so decided to knock it back to Jessie Lite and install your main version - bluetooth works perfectly but now Airport drops after 2-3 mins! So am now back on Stretch, might as well be on the latest and greatest.

Install is completely vanilla, only setup wifi and enabled SSH. Ran `sudo apt-get update` and `sudo apt-get upgrade` before your installer scripts.

`aplay -l` gives:

```
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
`asound.conf` has the following:

```
pcm.pulse {
    type pulse
}
ctl.pulse {
    type pulse
}

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}
```

`ps ax | grep pulse` gives:

`  330 ?        S<l    0:13 /usr/bin/pulseaudio --system --disallow-exit --disallow-module-loading=0 --daemonize --log-target=syslog --high-priority
  948 pts/0    S+     0:00 grep --color=auto pulse`

From researching i'm guessing I should be seeing something from `sudo pactl list sink-inputs` when a bluetooth device is connected and playing audio, but that gives nothing. Have tried from my iPhone and Macbook Pro - neither works, the music "plays" but no audio is heard. I'm absolutely stumped, any advice greatly appreciated!

randytsuch, 2018-02-19:
I installed stretch lite, and like the previous person, can't get it to work via bluetooth with an iphone.  It pairs, but I get no sound.  Works fine with my android phone after I turned off internal wifi.

wlemmers, 2018-02-20:
After clean installation it showed similar behavior as described above.
It turned out that Stretch lite on rpi0 used the wrong sink index. 
I am using the default alsa with a single card.

Check syslog: pulseaudio[310]: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "source=bluez_source.E0_99_71_DA_B1_D3.a2dp_source **sink=0**"): initialization failed.

Fixed it by changing /usr/local/bin/bluez-udev:  **audio_sink=1** , was 0

After that it works fine from Android or iPad.

OhmniD, 2018-02-20:
@wlemmers thanks, blanked it again and that fixes it - performance seems to skip/jump compared to the manual setup I tried above though.

George-L, 2018-02-21:
Hello. I just installed this on my rpi3 runing the latest version of raspbian stretch and i have 2 problems. the bluetooth wont send any sound until i reconnect 2 times after reboot and after that the sound has cracking sounds in the background. The wi-fi is disabled and i chose bluetooth only install with no sound card.

mbrebion, 2018-02-26:
I've try to install on an a stretch sd card which on which I've tried to setup properly blue-asla (without success). I' think that this lead to conflict so that the installation of your script failed (bluetooth script).
After reinstalling stretch and sudo apt-get update / upgrade, a new install ran perfectly
Thanks again !

So an advice for those who encounters problem with stretch : start from a freshly installed stretch image

BaReinhard, 2018-03-03:
Blue-alsa has a completely different approach to bluetooth playback. I haven't had success finding an auto connect functionality with it as of yet (not much dev time with current work/school schedule). This project works best (maybe only?) with a fresh install. @mbrebion .

@George-L when you say cracking sounds, does the playback stop or just poor quality? Poor quality is an issue with the builtin sound. As such I suggest even using a cheap $6 sound card from sabrent, [here](https://www.amazon.com/Sabrent-External-Adapter-Windows-AU-MMSA/dp/B00IRVQ0F8/ref=sr_1_4?s=electronics&ie=UTF8&qid=1520115064&sr=1-4&keywords=sabrent+usb+sound&dpID=51QY728wxCL&preST=_SY300_QL70_&dpSrc=srch)

BaReinhard, 2018-03-03:
The functionality of this thread has served its purpose for the initial testing. I believe continuing this thread will ultimately cause more headache than help. As such for anyone with existing/new problems please open a new issue (relevant to your issue)

Thanks for all the feedback so far!

P.S. Pull Requests are always welcome!

closed 2018-03-03 by BaReinhard

comarius, 2018-04-01:
Hi, How do I use this ? I understood snapcast in one browse, but this, what does it do. 
Thank you.

BaReinhard, 2018-04-02:
@comarius thanks for pointing this out. It should be addressed in the README.md but it is not I’ll be sure to add it soon.

Basically, this is an installation script that setups up the raspberry pi as an audio receiver. Which allows you to cast music to connected speakers via AirPlay or Bluetooth. The Snapcast addition to the project allows for the same functionality with multiroom support.

comarius, 2018-04-02:
Thank you. Got it.
Please close.

shamitoH, 2018-10-24:
So here's my feedback. 
I'm using raspberry pi 3 with Raspbian Stretch.
Installed following the instructions above, so at initialization after first reboot get the message 

```
**[FAILED]** Failed to start LSB: Makes Bluetooth discoverable and connectable to 0000...
 See "systemctl status bluetooth-agent.service" for details.
```
getting the following:

```
pi@raspberrypi:~ $ systemctl status bluetooth-agent.service
● bluetooth-agent.service - LSB: Makes Bluetooth discoverable and connectable to 0000
   Loaded: loaded (/etc/init.d/bluetooth-agent; generated; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-10-24 01:27:28 UTC; 4min 4s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 409 ExecStart=/etc/init.d/bluetooth-agent start (code=exited, status=1/FAILURE)

Oct 24 01:27:26 raspberrypi systemd[1]: Starting LSB: Makes Bluetooth discoverable and connectable to 0000...
Oct 24 01:27:26 raspberrypi bluetooth-agent[409]: setting bluetooth discoverable
Oct 24 01:27:28 raspberrypi sudo[430]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/hciconfig hci0 piscan
Oct 24 01:27:28 raspberrypi sudo[430]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 24 01:27:28 raspberrypi bluetooth-agent[409]: Can't get device info: No such device
Oct 24 01:27:28 raspberrypi sudo[430]: pam_unix(sudo:session): session closed for user root
Oct 24 01:27:28 raspberrypi systemd[1]: bluetooth-agent.service: Control process exited, code=exited status=1
Oct 24 01:27:28 raspberrypi systemd[1]: Failed to start LSB: Makes Bluetooth discoverable and connectable to
Oct 24 01:27:28 raspberrypi systemd[1]: bluetooth-agent.service: Unit entered failed state.
Oct 24 01:27:28 raspberrypi systemd[1]: bluetooth-agent.service: Failed with result 'exit-code'.
```
The raspberry was visible in other devices but unable to connect. The issue can be solved by restarting the service, but you must do it every time the raspberry is restarted.

I realized that bluetooth-agent.service starts before Bluetooth service does. So edited the init.d file for bluetooth-agent. Adding some seconds (5 works fine for me) in the sleep line at the start header, there would be time to bluetooth service to start. So here's my /etc/init.d/bluetooth-agent:

```
#!/bin/sh -e
### BEGIN INIT INFO
# Provides:             bluetooth-agent
# Required-Start:       $remote_fs $syslog bluetooth pulseaudio
# Required-Stop:        $remote_fs $syslog
# Default-Start:        2 3 4 5
# Default-Stop:         0 1 6
# Short-Description:    Makes Bluetooth discoverable and connectable to 0000
# Description:          Start Bluetooth-Agent at boot time.
### END INIT INFO
#! /bin/sh
# /etc/init.d/bluetooth-agent
USER=root
HOME=/root
export USER HOME
case "$1" in
    start)
        echo "setting bluetooth discoverable"
        sleep 5
        sudo hciconfig hci0 piscan
        start-stop-daemon -S -x /usr/local/bin/simple-agent.autotrust &
        echo "bluetooth-agent started"
        ;;
    stop)
        echo "Stopping bluetooth-agent"
        start-stop-daemon -K -x /usr/local/bin/simple-agent.autotrust
        ;;
    *)
        echo "Usage: /etc/init.d/bluetooth-agent {start|stop}"
        exit 1
        ;;
esac
exit 0

```

I'm not an english speaker, so sorry if I use inappropriate language or inconsistencies.