BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #103: No sound via BT in Volumio

lellmauer, 2018-01-09:
Hi there, I installed Volumio 2.348 yesterday on an RPi 3. Afterwards I followed the steps you described in the Volumio forum (https://volumio.org/forum/volumio-bluez-alsa-a2dp-bluetooth-support-t6130-20.html). Installation went well and I could connect my phone to the Pi after it. When playing music on the phone, no audio is played on the Pi (I have to admit though, that I am not sure what to set in Volumio to get the BT audio, so this might also cause the problem). You also mentioned that I "may need to configure what pulseaudio input sink goes to what output sink." - could you explain that? Sorry for the stupid questions and thanks in advance!

BaReinhard, 2018-01-09:
No worries I should have done a better job with the details. Two questions:

1. Are you running an i2s dac? 
2. Did you install from the volumio plugin or through ssh? (ssh seems to be better, as i had to do a few hacky things to get the plugin to work, and I need to look more into who plugins are run)

lellmauer, 2018-01-09:
Thanks for the quick reply. Yes, I am using various i2s DACs for testing. I have a Mamboberry LS, Hifiberry, Suptronics X5000 (Hifiberry compatible), and a small IQAudio DAC for the Zero, can't remember the name. I did the installation via SSH, I didn't even know there was a plugin. So after writing the Volumio image to the SD card, I followed the steps you described:

git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
git checkout volumio-fix
sudo ./install.sh

I chose option #4 and a device name. Installation took some time, but completed succesfully. Now I can connect my phone to the Pi, but this is where I am stuck at the moment. How am I supposed to play back files via BT? Just start the playback on the external device and Volumio will detect it automatically, or do I have to select a certain menu option?

Would be great if you had any hints for troubleshooting. I am not really a Linux expert, so I am afraid I cannot contribute codewise, but if you need someone for testing certain things, just let me know.

Best regards,

Lars

BaReinhard, 2018-01-09:
Thanks Lars, for now the use of Bluetooth isn’t directly integrated with volumio it should just start playing there are some things I’ll need to be able to diagnose the issue:

Make sure you’re device is connected to the Pi and you’re playing some sort of music.

Then run the following commands

`sudo pactl list sinks`
`sudo pactl list sink-inputs`

And reply with the outputs of both.

lellmauer, 2018-01-14:
Hi Brett,

sorry it took me so long to reply, it was my daughter's birthday yesterday and I was caught up in family stuff the whole week.

Anyway, here's the output:

Sinks

Sink #0
State: RUNNING
Name: alsa_output.platform-soc_audio.analog-stereo
Description: bcm2835 ALSA Analog Stereo
Driver: module-alsa-card.c
Sample Specification: s16le 2ch 44100Hz
Channel Map: front-left,front-right
Owner Module: 1
Mute: no
Volume: front-left: 65536 / 100% / 0.00 dB, front-right: 65536 / 100% / 0.00 dB
balance 0.00
Base Volume: 56210 / 86% / -4.00 dB
Monitor Source: alsa_output.platform-soc_audio.analog-stereo.monitor
Latency: 93257 usec, configured 100000 usec
Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY
Properties:
alsa.resolution_bits = "16"
device.api = "alsa"
device.class = "sound"
alsa.class = "generic"
alsa.subclass = "generic-mix"
alsa.name = "bcm2835 ALSA"
alsa.id = "bcm2835 ALSA"
alsa.subdevice = "0"
alsa.subdevice_name = "subdevice #0"
alsa.device = "0"
alsa.card = "0"
alsa.card_name = "bcm2835 ALSA"
alsa.long_card_name = "bcm2835 ALSA"
alsa.driver_name = "snd_bcm2835"
device.bus_path = "platform-soc:audio"
sysfs.path = "/devices/platform/soc/soc:audio/sound/card0"
device.string = "hw:0"
device.buffering.buffer_size = "17640"
device.buffering.fragment_size = "1760"
device.access_mode = "mmap"
device.profile.name = "analog-stereo"
device.profile.description = "Analog Stereo"
device.description = "bcm2835 ALSA Analog Stereo"
alsa.mixer_name = "Broadcom Mixer"
module-udev-detect.discovered = "1"
device.icon_name = "audio-card"
Ports:
analog-output: Analog Output (priority: 9900)
Active Port: analog-output
Formats:
pcm

Sink #1
State: SUSPENDED
Name: alsa_output.platform-soc_sound.analog-stereo
Description: snd_rpi_hifiberry_dac Analog Stereo
Driver: module-alsa-card.c
Sample Specification: s16le 2ch 44100Hz
Channel Map: front-left,front-right
Owner Module: 2
Mute: no
Volume: front-left: 65536 / 100% / 0.00 dB, front-right: 65536 / 100% / 0.00 dB
balance 0.00
Base Volume: 65536 / 100% / 0.00 dB
Monitor Source: alsa_output.platform-soc_sound.analog-stereo.monitor
Latency: 0 usec, configured 0 usec
Flags: HARDWARE DECIBEL_VOLUME LATENCY
Properties:
alsa.resolution_bits = "16"
device.api = "alsa"
device.class = "sound"
alsa.class = "generic"
alsa.subclass = "generic-mix"
alsa.name = ""
alsa.id = "HifiBerry DAC HiFi pcm5102a-hifi-0"
alsa.subdevice = "0"
alsa.subdevice_name = "subdevice #0"
alsa.device = "0"
alsa.card = "1"
alsa.card_name = "snd_rpi_hifiberry_dac"
alsa.long_card_name = "snd_rpi_hifiberry_dac"
alsa.driver_name = "snd_soc_hifiberry_dac"
device.bus_path = "platform-soc:sound"
sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
device.string = "hw:1"
device.buffering.buffer_size = "17640"
device.buffering.fragment_size = "1764"
device.access_mode = "mmap"
device.profile.name = "analog-stereo"
device.profile.description = "Analog Stereo"
device.description = "snd_rpi_hifiberry_dac Analog Stereo"
module-udev-detect.discovered = "1"
device.icon_name = "audio-card"
Ports:
analog-output: Analog Output (priority: 9900)
Active Port: analog-output
Formats:
pcm

Sink Inputs

Sink Input #0
Driver: module-loopback.c
Owner Module: 16
Client: n/a
Sink: 0
Sample Specification: s16le 2ch 43236Hz
Channel Map: front-left,front-right
Format: pcm, format.sample_format = "\"s16le\"" format.rate = "44100" format.channels = "2" format.channel_map = "\"front-left,front-right\""
Corked: no
Mute: no
Volume: front-left: 65536 / 100% / 0.00 dB, front-right: 65536 / 100% / 0.00 dB
balance 0.00
Buffer Latency: 98297 usec
Sink Latency: 94470 usec
Resample method: speex-fixed-1
Properties:
media.role = "abstract"
module-stream-restore.id = "sink-input-by-media-role:abstract"
media.name = "Loopback from Moto G (4)"
media.icon_name = "audio-card-bluetooth"

Hope this helps you to help me ;-)

Thanks for your support!

Best regards,
Lars

BaReinhard, 2018-01-14:
It’s all good @lars I hope your daughter had a nice birthday. How old did she turn?

In regards to the playback issue, what you can do to solve this issue is edit `/usr/local/bin/bluez-udev` On the second line there is something that says `audio_sink=0` change that to `audio_sink=1` next time you connect it should automatically play to the hifiberry amp.

lellmauer, 2018-01-14:
Thanks, that worked fine. Music playback from my phone is now streamed on the RPi. 😃

Playback is stuttering heavily though, even if the phone is lying right next to the Pi. I tried playback from music stored on the phone via Jet Audio Player as well as streamed vis Spotify app.

Any hints on how to improve this? Or is is better to treat this in a new issue, as this one is solved.

Again, thanks for your help, if you need me to test anything, let me know.

Cheers,
Lars

lellmauer, 2018-01-14:
Oh, nearly forgot.She turned 18, therefore it had to be somewhat bigger this time... 😉

Everything went fine, and now proud daddy is looking forward to having more time for hobbying in the future 😜

BaReinhard, 2018-01-14:
Haha very nice, I could imagine 18 is a big deal. Luckily my daughter just turned 2 so I have some time before the big 18.

That’s odd that playback is choppy, a start would be to take a look at the syslog,

`cat /var/lightshowpi/syslog`
As well as running

`pulseaudio —version`

I would first try restarting the pi and try again before the previous commands.

frmelle, 2018-01-14:
Hello, same thing here, Volumio on Raspberry pi 3 with Pifi DAC+ and with the audio_sink=1 I do have sound but it doesn't work properly. It stops about twice a second ... Streaming is no working flawlessly :-(

Do you need anything from me to troubleshoot ?

BaReinhard, 2018-01-14:
Try running the commands I suggested the post right above your comment.

BaReinhard, 2018-01-14:
Also, are these connected to the internet via WiFi or Ethernet?

frmelle, 2018-01-15:
My RPi3 is connected via Wifi to the internet. You think that if I turn off
wifi, it will work better ?

frmelle, 2018-01-15:
pulseaudio -version
pulseaudio: invalid option -- 'e'
E: [pulseaudio] main.c: Failed to parse command line.

cat /var/lightshowpi/syslog
cat: /var/lightshowpi/syslog: No such file or directory

BaReinhard, 2018-01-15:
@frmelle 
Unfortunately, streaming bluetooth and using wifi is a bug with the drivers that Raspbian uses for the broadcom chipset. It been reported to the RPI foundation, but it doesn't seem like it's that high a priority at the moment. A current workout around is using a ethernet connection or an external wifi dongle to handle connection to wifi. You can take a look at the issue here: https://github.com/raspberrypi/linux/issues/1402

For pulseaudio it should be
 
```
pulseaudio --version
```

at the log statement should have been

```
cat /var/log/syslog
```

Sorry I was commenting from my phone it must have autocorrected to lightshowpi, kinda odd.

frmelle, 2018-01-16:
Here are the logs 👍

```
volumio@radiopi:~$ pulseaudio --version
pulseaudio 6.0
```
But I don't have any syslog

```
volumio@radiopi:~$ ls /var/log/
boot.log  btmp  volumio.log  wtmp
```
`[   12.908951] brcmfmac: power management disabled
[   13.832792] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.883951] brcmfmac: power management disabled
[   14.351514] ip_tables: (C) 2000-2006 Netfilter Core Team
[   14.682841] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
[   17.082685] random: crng init done
[   17.735239] pcm512x 1-004d: No SCLK, using BCLK: -2
[   17.740300] pcm512x 1-004d: No SCLK, using BCLK: -2
[   17.817982] pcm512x 1-004d: No SCLK, using BCLK: -2
[   17.822923] pcm512x 1-004d: No SCLK, using BCLK: -2
[   17.830107] pcm512x 1-004d: No SCLK, using BCLK: -2
[   17.862988] pcm512x 1-004d: No SCLK, using BCLK: -2
[   18.595442] Bluetooth: Core ver 2.22
[   18.595516] NET: Registered protocol family 31
[   18.595522] Bluetooth: HCI device and connection manager initialized
[   18.595545] Bluetooth: HCI socket layer initialized
[   18.595559] Bluetooth: L2CAP socket layer initialized
[   18.595593] Bluetooth: SCO socket layer initialized
[   18.659573] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   18.659583] Bluetooth: BNEP filters: protocol multicast
[   18.659602] Bluetooth: BNEP socket layer initialized
[   19.571194] uart-pl011 3f201000.serial: no DMA platform data
[   24.056256] Bluetooth: HCI UART driver ver 2.3
[   24.056269] Bluetooth: HCI UART protocol H4 registered
[   24.056274] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   24.056457] Bluetooth: HCI UART protocol Broadcom registered
[   24.110264] Bluetooth: RFCOMM TTY layer initialized
[   24.110294] Bluetooth: RFCOMM socket layer initialized
[   24.110315] Bluetooth: RFCOMM ver 1.11
[   27.824031] brcmfmac: brcmf_cfg80211_stop_ap: setting INFRA mode failed -7
[   45.604486] FS-Cache: Netfs 'cifs' registered for caching
[   45.605310] Key type cifs.spnego registered
[   45.605332] Key type cifs.idmap registered
[   49.346617] pcm512x 1-004d: No SCLK, using BCLK: -2
[  575.108083] input: 94:8B:C1:78:5E:79 as /devices/virtual/input/input0
[  576.823720] pcm512x 1-004d: No SCLK, using BCLK: -2
[  587.133507] pcm512x 1-004d: No SCLK, using BCLK: -2
`

BaReinhard, 2018-01-16:
Ok all looks good. Can you try connecting the pi to an Ethernet cable and see if the stutter continues?

frmelle, 2018-01-16:
I disabled wifi and all is working perfectly !

So I went to autostart (/etc/rc.local) at launch this script. I works perfectly :-)

```
#!/bin/bash

# enable wifi first to always gain access on boot
ifconfig wlan0 up

# When BT connexion started : Wifi OFF
# When BT connexion stopped : Wifi ON
while true
do
RES=`inotifywait -q -e CREATE,DELETE /dev/input/`
case "$RES" in
"/dev/input/ DELETE event0")
ifconfig wlan0 up
;;
"/dev/input/ CREATE event0")
ifconfig wlan0 down
;;
esac
done &
```

lellmauer, 2018-01-17:
Same here, when connected via Ethernet with WiFi disabled, music playback works fine. Since I need the wireless connection, I will give it another try with an external dongle tonight. Will get back with the results.

silelmot, 2018-01-21:
i read about the problem, that onboard bluetooth and wireless still dont work flawlessly on rapsberry 3b, so i use a dongle for wifi now and this works great.
hope this will be fixed soon.
but as i see, this is nothing BaReinhard can do. :(

silelmot, 2018-01-21:
as a site note: with the new settings in bluez-udev: audio_sink=1 the soudn doesnt work for me with onboard-sound anymore. i changed it back to 0 and it works, 
@BaReinhard  are you able to get this automated within your install-script?

BaReinhard, 2018-01-21:
@silelmot yeah that should be an easy fix, it’ll have to wait until I get some time. The reason it doesn’t work with 1 is 1 means it’s expecting more than one audio source , I.e. a dac or usb sound card.

D10Z, 2018-02-13:
For both BT and Airplay services installed, if you use the disable_wifi_on_bt.sh script and you connect to the BT, the WiFi will be disabled. 
But the shairport-sync won't have the time to broadcast the end of its service.
So, if you use an iOS device, you would see for a long time the AirPlay service enabled although it is not.
I modified the script : 
```
#!/bin/bash`

while true
do
RES=`inotifywait -q -e CREATE,DELETE /dev/input/`
case "$RES" in
"/dev/input/ DELETE event0")
ifconfig wlan0 up
systemctl start shairport-sync
;;
"/dev/input/ CREATE event0")
systemctl stop shairport-sync
sleep 1
ifconfig wlan0 down
;;
esac
done &

Now on a BT connection, the shairport-sync service is disabled and the enable again when WiFi comes up.
```

frmelle, 2018-02-13:
Ok, so it is quite strange that on a Volumio installation (I retried
yesterday with the latest image 12-02-2018 V2.362) and I still haven't any
BT connection showing up ...

The thing is that as D10Z when I install your script upon a fresh raspbian
install, it works ... I can't figure out what I am missing :-(

BaReinhard, 2018-02-17:
I haven’t checked recently if there was a new image for volumio. If they updated their pre baked image there may be some issues with the installation as it was already pretty fragile with the older image.

George-L, 2018-02-21:
Hello. I just installed this on my rpi3 runing the latest version of raspbian stretch and i have 2 problems. the bluetooth wont send any sound until i reconnect and after that the sound has cracking sounds in the background. The wi-fi is disabled and i chose bluetooth only install with no sound card.

flasche1, 2018-04-19:
Does anyone got it work on Volumio 2.389 with raspberry pi 3b+ ?
Or do i have to use raspbian?
I only need working bluetooth streaming with auto connect.

s1m00, 2018-04-30:
Hello flasche1, same thing happened to me.
I was able to complete installation following a couple of tips spread in blog:
- edit bt_pa_config.sh with "sudo nano" and change every "sudo" command in "sudo -S";
- delete "exc" tag before new "sudo - S chmod XXX" (where XXX are numbers of read-write permissions) and "sudo - S patch". Someone suggested to remove all, but in my case worked in this way
- launch again installation with "sudo ./install.sh"
- at a certain point (when first "sudo -S" command is met installing "volume-watcher.py") it will stop and seems to freeze. You have to type in your password for new "sudo -S" command and ENTER
- Installation is now able to end without errors...in my case.
I hope this will help you.

flasche1, 2018-05-01:
Thanks for your help @s1m00.
It worked after I changed only all "sudo" to "sudo -S", no deleting of a "exc" was needed.
Best regards,
Andi