BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #104: Audio Issues

mdrons, 2018-01-10:
Something changed when I tried to switch to the headphone jack.   My initial setup worked with sound over hdmi.   Now I am seeing these error messages in the syslog.  Not sure where to being to look.  
Thanks for any help.

```Jan 10 16:38:32 raspberrypi pulseaudio[448]: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set.
Jan 10 16:38:32 raspberrypi pulseaudio[448]: [pulseaudio] main.c: Running in system mode, forcibly disabling SHM mode.
Jan 10 16:38:32 raspberrypi pulseaudio[484]: [pulseaudio] main.c: OK, so you are running PA in system mode. Please make sure that you actually do want to do that.
Jan 10 16:38:32 raspberrypi pulseaudio[484]: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
Jan 10 16:38:33 raspberrypi pulseaudio[484]: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.config/pulse/cookie': No such file or directory
Jan 10 16:38:33 raspberrypi pulseaudio[484]: [pulseaudio] authkey.c: Failed to load authentication key '/var/run/pulse/.config/pulse/cookie': No such file or directory
Jan 10 16:38:33 raspberrypi pulseaudio[484]: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.pulse-cookie': No such file or directory
Jan 10 16:38:33 raspberrypi pulseaudio[484]: [pulseaudio] authkey.c: Failed to load authentication key '/var/run/pulse/.pulse-cookie': No such file or directory
Jan 10 16:49:02 raspberrypi shairport-sync[580]: Waiting to connect to bluetoothd...#015#033[0;94m[bluetooth]#033[0m# discoverable off
Jan 10 16:49:02 raspberrypi shairport-sync[580]: No default controller available
Jan 10 16:49:02 raspberrypi bluetoothctl[710]: Source ID 14 was not found when attempting to remove it
Jan 10 16:49:02 raspberrypi shairport-sync[580]: #033[0;94m[bluetooth]#033[0m# #015Waiting to connect to bluetoothd...#015                                   #015Failed to create secure directory (/home/shairport-sync/.config/pulse): No such file or directory
Jan 10 16:49:02 raspberrypi shairport-sync[580]: Failed to load cookie file from cookie: No such file or directory
Jan 10 16:49:02 raspberrypi shairport-sync[580]: Failed to create secure directory (/home/shairport-sync/.config/pulse): No such file or directory
Jan 10 16:49:02 raspberrypi shairport-sync[580]: Failed to load cookie file from cookie: No such file or directory
Jan 10 16:49:02 raspberrypi shairport-sync[580]: ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied
Jan 10 16:49:02 raspberrypi pulseaudio[484]: [pulseaudio] protocol-native.c: Denied access to client with invalid authentication data.
Jan 10 16:49:02 raspberrypi shairport-sync[580]: amixer: Control default open error: Connection refused
Jan 10 16:49:25 raspberrypi shairport-sync[580]: Waiting to connect to bluetoothd...#015#033[0;94m[bluetooth]#033[0m# discoverable on
Jan 10 16:49:25 raspberrypi shairport-sync[580]: No default controller available
Jan 10 16:49:25 raspberrypi bluetoothctl[715]: Source ID 13 was not found when attempting to remove it
Jan 10 16:49:25 raspberrypi shairport-sync[580]: #033[0;94m[bluetooth]#033[0m# #015Waiting to connect to bluetoothd...#015                                   #015Failed to create secure directory (/home/shairport-sync/.config/pulse): No such file or directory
Jan 10 16:49:25 raspberrypi shairport-sync[580]: Failed to load cookie file from cookie: No such file or directory
Jan 10 16:49:25 raspberrypi shairport-sync[580]: Failed to create secure directory (/home/shairport-sync/.config/pulse): No such file or directory
Jan 10 16:49:25 raspberrypi shairport-sync[580]: Failed to load cookie file from cookie: No such file or directory
Jan 10 16:49:25 raspberrypi shairport-sync[580]: ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied
Jan 10 16:49:25 raspberrypi shairport-sync[580]: amixer: Control default open error: Connection refused
Jan 10 16:49:25 raspberrypi pulseaudio[484]: [pulseaudio] protocol-native.c: Denied access to client with invalid authentication data.
```

BaReinhard, 2018-01-10:
How did you try to set output to audio jack?

mdrons, 2018-01-11:
raspi-config to change the audio out.   I also used it to change it back to hdmi.

Thanks,
Mike

BaReinhard, 2018-01-11:
Ok, based on your error log, what does 

`cat /etc/asound.conf`

Output?

mdrons, 2018-01-11:
cat /etc/asound.conf
pcm.pulse {
    type pulse
    card 0
}
ctl.pulse {
    type pulse
    card 0
}

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}

BaReinhard, 2018-01-11:
Change the asound.cond file to be the following:

```
pcm.pulse {
type pulse
}
ctl.pulse {
type pulse
}

pcm.!default {
type hw
card 0
}
ctl.!default {
type hw
card 0
```

mdrons, 2018-01-11:
I reinstalled and the issue is fixed.  Not sure what got changed.   Thanks

closed 2018-01-11 by mdrons

BaReinhard, 2018-01-11:
Glad it worked!  Could be the recent commits I made, but glad it worked.