BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #107: No audio after install

XavierMeunier, 2018-01-15:
Hi,

I've just installed your awesome lib, I connect through BT and I choose my RPi as Airplay device but no sound is played.

I'm on Jessie (not LITE).
I've tried to play some music via Youtube directly on the RPI via graphic interface but no sound either

Here is my install_choices:
```
Bluetooth = y
Airplay= n
AP = n
Kodi = n
Lirc = n
SoundCardInstall = n
GMedia = n
SnapCast = n
```

BaReinhard, 2018-01-15:
I would try changing the `/etc/asound.conf` to fix this. 

First look at your devices via `aplay -l`

Then whatever your output device of choice is change your `/etc/asound.conf` to the following:

```
pcm.!default{
   type hw
   card CARD_NUMBER_FROM_APLAY
}
ctl.!default{
    type hw
    card CARD_NUMBER_FROM_APLAY
}
pcm.pulse{
    type pulse
}
ctl.pulse{
    type pulse
}
```

XavierMeunier, 2018-01-16:
In the mean time, I re-installed with a different config: Car Installation ; and I met a fatal error on kodi config: 

```
Fatal error encountered: 'sed -i s/ENABLED=0/ENABLED=1/ /etc/default/kodi' failed
Fatal error encountered: ./kodi_config.sh failed
```

FYI - When I launch kodi_install.sh , everything is fine: `Dependency kody already met`

Strangely, I can still connect to the RPi & select it as audio airplay.
I set your asound config but it was already in place (but in a different order).

`aplay -l`
```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
`asound.conf`
```
pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}
pcm.pulse {
    type pulse
}
ctl.pulse {
    type pulse
}
```

XavierMeunier, 2018-01-16:
**It works.** 👍

I had a pHat DAC available & tried to set it up with the current installation and it worked instantly.
💡It could be a possible sound card to add in the install process. 💡

The problem probably came from my output sound settings as you pointed it out (I still don't exactly know where ... but who cares).

Thanks for your awesome work & your time at helping me.

closed 2018-01-16 by XavierMeunier

BaReinhard, 2018-01-17:
Glad it’s working!!