BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #112: Pi Zero W not seen as BT output

frmelle, 2018-01-25:
Hello after a successful install on Raspberry Pi 3 with pifiberry DAC+, I tried yesterday to install it on a Volumio fresh install on Raspberry pi Zero W with USB audio dongle.

First, I disabled the BCM2835 chip adding the line blacklist snd_bcm2835 to alsa-blacklist.conf.

All works fine in Volumio, I do have the sound output to USB audio out.

After installing Super-Simple-Raspberry-Pi-Audio-Receiver (choice 4). I saw the Pi Zero W as BT and managed to connect to it. The sound was choppy so I installed the script to watch BT connexion and disable wifi when  BT device is connected.

But upon reboot and even if comment the script call in /etc/rc.local, I can't see the Pi Zero W ... I have rebooted several times, no more luck.

```
pulseaudio --version
pulseaudio 6.0
```

```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.65+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1056 Fri Nov 24 13:53:32 GMT 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x1d800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 126976
[    0.000000] free_area_init_node: node 0, pgdat c0916df0, node_mem_map dd3a4000
[    0.000000]   Normal zone: 1116 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:56:89:B4 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  splash quiet plymouth.ignore-serial-consoles dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0xF dwc_otg.nak_holdoff=1 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfile=/volumio_current.sqsh elevator=noop rootwait bootdelay=5 logo.nologo vt.global_cursor_default=0 loglevel=0
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 470852K/507904K available (5958K kernel code, 491K rwdata, 1948K rodata, 396K init, 725K bss, 28860K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
    lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc05d9d68   (5960 kB)
      .init : 0xc0843000 - 0xc08a6000   ( 396 kB)
      .data : 0xc08a6000 - 0xc0920f28   ( 492 kB)
       .bss : 0xc0920f28 - 0xc09d6628   ( 726 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000060] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000148] bcm2835: system timer (irq = 27)
[    0.000618] Console: colour dummy device 80x30
[    0.000642] console [tty1] enabled
[    0.000668] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060303] pid_max: default: 32768 minimum: 301
[    0.060729] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060745] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062043] Disabling memory control group subsystem
[    0.062181] CPU: Testing write buffer coherency: ok
[    0.062240] ftrace: allocating 21720 entries in 64 pages
[    0.177934] Setting up static identity map for 0x8200 - 0x8238
[    0.179825] devtmpfs: initialized
[    0.189393] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.189803] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.189829] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.191086] pinctrl core: initialized pinctrl subsystem
[    0.192497] NET: Registered protocol family 16
[    0.195206] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.204845] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[... 179 lines cut ...]
[   23.750344] brcmfmac: power management disabled
[   25.779799] ip_tables: (C) 2000-2006 Netfilter Core Team
[   26.430784] nf_conntrack version 0.5.0 (8192 buckets, 32768 max)
[   46.458466] Bluetooth: Core ver 2.22
[   46.458641] NET: Registered protocol family 31
[   46.458649] Bluetooth: HCI device and connection manager initialized
[   46.458671] Bluetooth: HCI socket layer initialized
[   46.458684] Bluetooth: L2CAP socket layer initialized
[   46.458737] Bluetooth: SCO socket layer initialized
[   46.750247] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   46.750260] Bluetooth: BNEP filters: protocol multicast
[   46.750290] Bluetooth: BNEP socket layer initialized
[   48.793754] uart-pl011 20201000.serial: no DMA platform data
[   64.454678] brcmfmac: brcmf_cfg80211_stop_ap: setting INFRA mode failed -7
[   67.220558] brcmfmac: power management disabled
```

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 5: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

frmelle, 2018-01-25:
I changed the order of usb sound device in /etc/modprobe.d/alsa-base.conf from 5 (set by volumio) to 0.

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I still have the same issue ...

frmelle, 2018-02-01:
Hello, I tried a fresh install in custom mode, juste BT and usb soundcard but still the Raspberry Pi Zero W isn't seen as BT device.
What commands can I do to spot what is not working ? What process are supposed to expose the pi as a BT sound device ? And where are the config and log files ?

Thank-you.

BaReinhard, 2018-02-05:
what OS is installed on the fresh install?

frmelle, 2018-02-05:
I installed latest Volumio 2 image. On a fresh Jessie install, your
installation works but then I am not able to setup up Volumio on top of a
raspbian Jessie Lite install ...

My goal is just to add Bluetooth connectivity to a Volumio install on
Raspberry Pi Zero W.

Thank-you for your help.

BTW, uname -a gives :

Linux concertboy 4.9.65+ #1056 Fri Nov 24 13:53:32 GMT 2017 armv6l GNU/Linux

D10Z, 2018-02-13:
Hi,

I'm testing your fantastic script on a RPi0W too.
My installation (Raspbian lite) is fresh (apt-get update and upgrade) and customed by your script (menu 6 - BT only or BT + AirPlay).

I'm encountering weird behaviours :
The RPi0 is starting well, no special dmesg error messages, the BT is detected by my smartphone, it can connect to it but :
- Sometimes (most of the time) the sound cannot be heard, although AirPlay sound works well.
- When the sound can be heard, I can connect, disconnect many times, it works perfectly.
- When I plug the same SD card OS in a RPi3, everything works fine every time.

I think the CPU load at start is not the same and that could affect the process start flow (order). 
I did some outputs (dmesg, systemctl status bluetooth* and pulseaudio), but they seem irrelevant.

I'm volonteer to do some tests and traces but I have to admit that I'm not familiar with Pulseaudio and Bluetooth neither.

frmelle, 2018-02-14:
OK, tonight I did some more digging on google and found an hint.

I added `hciattach /dev/ttyAMA0 bcm43xx 921600 flow -` to `/etc/rc.local` and after reboot, the raspberry Pi ZeroW is seen as a BT device.

Unfortunately when I try to connect to it it doesn't work at all :(

Any hint how to troubleshoot BT connection ?

frmelle, 2018-02-14:
I found tonight that if I manually issue the command `sudo /etc/init.d/bluetooth-agent start` after Volumio started (earing the starting sound), I see the Pi ZeroW as BT device and I can connect to it.

Unfortunately, I am unable to stream any sound to it ... Do you know what commands I can issue to check why streaming audio over BT isn't working ?

frmelle, 2018-02-15:
Having slept on it, I set up this morning before leaving home a complete workaround.

So to have this script work to add BT connectivity as a speaker to my Volumio 2 install, here is what I did. Consider that I use an USB dongle as audio out device as default jack output is very bad on Rpi.

- Install and configure Volumio 2 on the Rpi 0w with the latest Volumio 2 image
- Install Super-Simple-Raspberry-Pi-Audio-Receiver-Install using the BT only choice (choice 4), note that I used the volumio-fix version on github (`git checkout volumio-fix`)
- Disable onboard audio :
`sudo nano /etc/modprobe.d/alsa-blacklist.conf`
- Enter the following line :
`blacklist snd_bcm2835`
- Save the file and reboot
`sudo reboot`
- Create a script that I called `bt_audio.sh`
```
#!/bin/bash

# Wait 3 minutes before starting the script
sleep 180

# Make Rpi Zero W seen as a BT device
/etc/init.d/bluetooth-agent stop
sleep 5
/etc/init.d/bluetooth-agent start

# Enable streaming audio to BT device
/etc/init.d/pulseaudio stop
sleep 5
/etc/init.d/pulseaudio start

# Wait 10 seconds before continuining the script
sleep 10

# When BT connexion started : Wifi OFF
# When BT connexion stopped : Wifi ON
while true
do
RES=`inotifywait -q -e CREATE,DELETE /dev/input/`case "$RES" in
"/dev/input/ DELETE event1")
ifconfig wlan0 up
;;
```
- Make the script executable
`chmod +x /home/volumio/bt_audio.sh`
- Add these lines to `/etc/rc.local`
```
# enable wifi first to always gain access on boot
ifconfig wlan0 up

# script to stop wifi when connected to BT
/home/volumio/bt_audio.sh &
```
- Reboot the Rpi 0w and 3 minutes after earing the Volumio startup sound, I am able to connect to my Rpi 0w as a BT speaker !
`sudo reboot`

Perhaps theses informations would help BaReinhard tweak its Volumio branch of this great script !