BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #114: Analog audio not working

EtienneMD, 2018-02-06:
Sorry if the solution seems obvious to everybody, as this is my first project with RPi audio. 
I've run the install script with option 4 (BT only) successfully. I'm able to pair my phone over bluetooth and play audio through HDMI. However, I'm trying to get the audio from the audio jack without success. I've tried forcing output with `raspi-config` and tried running `sudo amixer cset numid=3 1` without success. I really don't know where to go from here. Will do some more research when I can.

BaReinhard, 2018-02-07:
Can you paste the contents of your `/etc/asound.conf` file

EtienneMD, 2018-02-07:
`asound.conf` did not exist, so I ran the install script again using custom install so I could select bluetooth and Soundcard 0 (no soundcard). This generated the proper file where it should be.

`pcm.pulse {
    type pulse
    card 0
}
ctl.pulse {
    type pulse
    card 0
}
pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}`

EtienneMD, 2018-02-07:
I am still faced with the same issue, sound will come out of HDMI no matter what and if HDMI is disconnected I get no sound at all.

EtienneMD, 2018-02-10:
Fixed with fresh raspbian install.

closed 2018-02-10 by EtienneMD

BaReinhard, 2018-02-13:
sorry we couldn't get this fixed with the existing install, but glad you have gotten it working!