BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #119: etc default kodi is missing?

abuonline, 2018-03-01:
Fatal error encountered: 'sed -i s/ENABLED=0/ENABLED=1/ /etc/default/kodi' failed
Fatal error encountered: ./kodi_config.sh failed

BaReinhard, 2018-03-03:
Which branch was this? Kodi has been deprecated as apart of this project since a recent breaking change and I haven't had the time to update the code base.

araufdogan, 2018-03-04:
I am getting this error too. Master branch, could you please fix? @BaReinhard

IncognitoJam, 2018-03-29:
This is still an issue. Following the installation instructions in the README on the master branch it attempts to install kodi.

```
pi@raspberrypi:~/ $ sudo apt-get update
pi@raspberrypi:~/ $ sudo apt-get install git
pi@raspberrypi:~/ $ git clone https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install.git
pi@raspberrypi:~/ $ cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install
pi@raspberrypi:~/Super-Simple-Raspberry-Pi-Audio-Receiver-Install $ sudo ./install.sh
```

gterras, 2018-04-17:
I would suggest using Github `releases` for external projects such as Kodi or Shairport (ex https://github.com/xbmc/xbmc/releases/tag/17.6-Krypton). This way SSRPIARI won't break randomly if master are modified. By the way your project is awesome !!

BaReinhard, 2018-06-30:
Hey Guys, thanks for the interest in the project. Unfortunately, due to time constraints I am unable to address current issues and debugging, maintenance on the shell scripts are hard and inefficient. As such I will be porting over these scripts to ansible playbooks. This will allow for easier choosing of what you want installed, less brittle code base, and better maintenance so that less people experience the errors.

Once I have feature parity I will be addressing documentation on installing with ansible. The goal here is to allow people to setup multiple devices at once by only issuing one command, additionally to allow multiple OS support with less debugging and maintenance, as well as allowing other developers more easily add functionality to this repo with simple yaml files as opposed to obfuscated shell scripts.

[image]