BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #120: Running su volumio -c ./bt_pa_config.sh... Failing.

leetwanker, 2018-03-04:
> 
> SSRPARI [Sun Mar  4 23:08:18 UTC 2018] Running su volumio -c ./bt_pa_config.sh...
> SSRPARI [Sun Mar  4 23:08:18 UTC 2018] Executing sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
> Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
> Fatal error encountered: su volumio -c ./bt_pa_config.sh failed

I've tried to figure out why it's failing but the script is beyond my skill. This is my 2nd try on a fresh install. Thanks.

freakyc, 2018-03-04:
I had the same thing and I think it's from using SSH when running it.  I opened bt_pa_config.sh in nano and did a replace on 'sudo' and replaced it with 'sudo -S'.  This will output the prompt to provide your password for sudo to the screen.

The file functions.sh has the 'exc' function and it's sending the output of the commands it's executing to /dev/null.  I mention that in case you're having a separate issue, or it helps someone else with troubleshooting.

leetwanker, 2018-03-05:
@freakyc Okay, thanks! That worked. I replaced every instance of `sudo` with `sudo -S` in install.sh, bt_pa_config.sh and removed the calls for `exc` as well and it went through without a hitch!

BaReinhard, 2018-03-05:
Thanks for this update all! I will look into making this change in the coming weeks.

If anyone would be so kind to make a pull request for this it would be greatly appreciated. Possibly a PR on the volumio branch would be optimal or find a way to use a sudo command function and replace sudo with it and inside the sudo function we could have it check whether its a volumio install or other and use the specified install script for this.

closed 2018-03-05 by BaReinhard

freakyc, 2018-03-06:
I actually completely missed the mention of volumio.  I had to do the above on an install of Raspbian Stretch.

I imagine this is going to be necessary for any of the scripts called by another script, when the second script uses sudo.  I'll try to dig in to it more and see if I can put something together for a PR.