BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #121: [stretch] Access Point

totti2, 2018-03-06:
environment: stretch lite, raspberry pi 3
install choices: bluetooth, airplay, snapserver and -client

In order to get the acces point mode running again in a stretch install, I just followed this how-to:
https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

What works:  
- dhcp-server
- ssh into the raspberry pi 3
- while connected to my router, I have internet access on client devices
- snapserver and snapclient functionality fully working (after some [tweaking](https://gist.github.com/totti2/41ed90feb1eb5838cc6a789d2b2dd5a7#synchronous-multi-room-sound-casting-of-bluetooth-airplay-line-in-and-spotitfy) - no advertisement intended, but hopefully helpful)

Experiencing bandwidth issues:
- when I connect my smartphone to the wifi, sound gets choppy (bluetooth sound-source)
- after adding extra snapclients, sound gets choppy, too

I will try the same setup with an extra wifi-dongle. Or...  
Someone with more than my noobish copy-paste-expertise can help me (, please!).

Lauszus, 2018-03-10:
I have worked around the sound issue on Bluetooth by automatically disabling wlan0 when a device is connect.

Simply copy the following script into ```/etc/init.d/bt-audio-disable-wlan0.sh```:

```bash
#!/bin/bash
### BEGIN INIT INFO
# Provides:          bt-audio-disable-wlan0
# Required-Start:
# Required-Stop:
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Disables wlan0 when a Bluetooth device is connected
# Description:       Disables wlan0 when a Bluetooth device is connected
### END INIT INFO

# sudo apt-get install inotify-tools

while true; do
    RES=$(/usr/bin/inotifywait -q -e CREATE,DELETE /dev/input/)
    case "$RES" in
        "/dev/input/ DELETE event0")
            ifconfig wlan0 up
        ;;
        "/dev/input/ CREATE event0")
            ifconfig wlan0 down
        ;;
    esac
done
```

Then run:

```bash
sudo apt-get install inotify-tools
sudo chmod +x /etc/init.d/bt-audio-disable-wlan0.sh
sudo update-rc.d bt-audio-disable-wlan0.sh defaults
sudo reboot
```

totti2, 2018-03-12:
Without wifi I cannot control the snapcast server. Plus, all airplay-devices will lose connection. I'd rather keep those functions. 

But thanks for your action-on device-change approach. One could use that for an audio signal, indicating when a new bt-device has connected.

Lauszus, 2018-03-14:
@totti2 then the best solution would probably be to use a separate dongle for WiFi. That is what I have done myself.

totti2, 2018-03-19:
Alright. Then maybe [this](https://gist.github.com/totti2/e59c1bbf5c2e5ea3c52dc36cf12c8276) would be a good start. dependencies are not corrected yet.

totti2, 2018-03-24:
I tried this with a dedicated dongle and the initial bandwidth issue is gone (more testing need for snapserver, though).

My smartphone doesn't get an IP when the Pi isn't connected to a LAN. The moment I plug in the ethernet-cable, I get an IP for my smartphone. If I had to guess, I would suspect the dhcpcd not to work properly with both with and without ethernet-connection.

OFF: When using the integrated bluetooth-device I experience a lot of hick-ups and interruptions.

totti2, 2018-04-02:
[Finished](https://gist.github.com/totti2/e59c1bbf5c2e5ea3c52dc36cf12c8276). If someone could please check this.
to sum up:
- needs an seperate wifi-dongle
- needs dnsmasq in dependencies.sh
- replaces isc-dhcp-server by dnsmasq
- forwards ipv4-traffic
- snapcast, airplay, spotify, ssh - all wifi-capabilites are working fine

totti2, 2020-09-08:
Should have been done some days ago..

closed 2020-09-08 by totti2