BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #122: Installation of package 'libjson0-dev' failed

coopersi, 2018-03-12:
hi,

just came across this install and sounds right up my street for a car install i have going on.

All was ok until i got this message on installation....

SSRPARI [Mon 12 Mar 22:09:33 UTC 2018] Installing libjson0-dev...
Fatal error encountered: Installation of package 'libjson0-dev' failed
Fatal error encountered: ./bt_pa_install.sh failed

it then closed the install at that point

this is for raspberry pi3 with everything updated.

vmRelaxed, 2018-03-18:
same problem! need help

stephenblender, 2018-04-04:
Same issue.

`Processing triggers for libc-bin (2.24-11+deb9u3) ...
SSRPARI [Wed Apr  4 09:33:11 PDT 2018] Running ./bt_pa_install.sh...
SSRPARI [Wed Apr  4 09:33:11 PDT 2018] Raspbian Stretch Found
SSRPARI [Wed Apr  4 09:33:11 PDT 2018] Checking libjson-c-dev...
dpkg-query: no packages found matching libjson-c-dev
SSRPARI [Wed Apr  4 09:33:11 PDT 2018] Installing libjson-c-dev...
Fatal error encountered: Installation of package 'libjson-c-dev' failed
Fatal error encountered: ./bt_pa_install.sh failed
pi@piguy:~/super-simple-raspberry-pi-audio-receiver-install $ ssrpari
`

stephenblender, 2018-04-04:
I tried again, 'sudo ./install.sh', and seemed to make progress. 

```
pi@piguy:~/super-simple-raspberry-pi-audio-receiver-install $ sudo ./install.sh

SSRPARI [Wed Apr  4 09:41:49 PDT 2018] Restoring Original Files...
SSRPARI [Wed Apr  4 09:41:49 PDT 2018] Restoring .profile to /home/pi//.profile
.profile
SSRPARI [Wed Apr  4 09:41:49 PDT 2018] File '.profile' has been previously backed up
SSRPARI [Wed Apr  4 09:41:49 PDT 2018] Select Your Install Options
SSRPARI 1. Install the Raspberry Pi Audio Receiver Car Installation
SSRPARI 2. Install the Raspberry Pi Audio Receiver Home Installation
SSRPARI 3. Install the Raspberry Pi Network Without Internet Installation (For teaching!)
SSRPARI 4. Install the Bluetooth Only Installation
SSRPARI 5. Install the Snapcast Installation (BETA), choose from Snapcast Server, Client, or Both (Requires Minor Configuration)
SSRPARI 6. Install a Custom Raspberry Pi Audio Receiver
Which installation would you like to choose? (1/2/3/4/5/6) : 2
Do you want all the Devices to use the same name? (y/n) : y
Device name: piplate
Do you want to use an AirPlay password? (y/n) : y
Device AirPlay password: itsaniceplace
SSRPARI 0. No Sound Card
SSRPARI 1. HifiBerry DAC Light
SSRPARI 2. HifiBerry DAC Standard/Pro
SSRPARI 3. HifiBerry Digi+
SSRPARI 4. Hifiberry Amp+
SSRPARI 5. Pi-IQaudIO DAC
SSRPARI 6. Pi-IQaudIO DAC+, Pi-IQaudIO DACZero, Pi-IQaudIO DAC PRO
SSRPARI 7. Pi-IQaudIO DigiAMP
SSRPARI 8. Pi-IQaudIO Digi+
SSRPARI 9. USB Sound Card
SSRPARI 10. JustBoom DAC and AMP Cards
SSRPARI 11. JustBoom Digi Cards
Which Sound Card are you using? (0/1/2/3/4/5/6/7/8/9/10/11) : 10
SSRPARI [Wed Apr  4 09:42:35 PDT 2018] Updating via Apt-Get
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages [56.8 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.2 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages [1,360 B]
Fetched 11.8 MB in 20s (578 kB/s)
Reading package lists...
SSRPARI [Wed Apr  4 09:43:03 PDT 2018] Upgrading via Apt-Get
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
SSRPARI [Wed Apr  4 09:43:07 PDT 2018] Running ./bt_pa_install.sh...
SSRPARI [Wed Apr  4 09:43:07 PDT 2018] Raspbian Stretch Found
SSRPARI [Wed Apr  4 09:43:07 PDT 2018] Checking libjson-c-dev...
dpkg-query: no packages found matching libjson-c-dev
SSRPARI [Wed Apr  4 09:43:08 PDT 2018] Installing libjson-c-dev...
SSRPARI [Wed Apr  4 09:43:21 PDT 2018] Checking autopoint...
dpkg-query: no packages found matching autopoint
SSRPARI [Wed Apr  4 09:43:21 PDT 2018] Installing autopoint...
SSRPARI [Wed Apr  4 09:43:53 PDT 2018] Checking pulseaudio-module-bluetooth...
SSRPARI [Wed Apr  4 09:43:53 PDT 2018] Installing pulseaudio-module-bluetooth...
SSRPARI [Wed Apr  4 09:44:18 PDT 2018] Checking python-dbus...
SSRPARI [Wed Apr  4 09:44:18 PDT 2018] Dependency python-dbus already met...
SSRPARI [Wed Apr  4 09:44:18 PDT 2018] Checking libltdl-dev...
dpkg-query: no packages found matching libltdl-dev
SSRPARI [Wed Apr  4 09:44:18 PDT 2018] Installing libltdl-dev...
SSRPARI [Wed Apr  4 09:45:04 PDT 2018] Checking pulseaudio...
SSRPARI [Wed Apr  4 09:45:04 PDT 2018] Dependency pulseaudio already met...
SSRPARI [Wed Apr  4 09:45:04 PDT 2018] Checking libtool...
SSRPARI [Wed Apr  4 09:45:04 PDT 2018] Dependency libtool already met...
SSRPARI [Wed Apr  4 09:45:04 PDT 2018] Checking intltool...
dpkg-query: no packages found matching intltool
[... 175 lines cut ...]
SSRPARI [Wed Apr  4 10:03:25 PDT 2018] Installing libpolarssl-dev...
SSRPARI [Wed Apr  4 10:03:46 PDT 2018] Checking libsoxr-dev...
dpkg-query: no packages found matching libsoxr-dev
SSRPARI [Wed Apr  4 10:03:46 PDT 2018] Installing libsoxr-dev...
SSRPARI SSRPARI [Wed Apr  4 10:04:05 PDT 2018] Executing cd /root
[Wed Apr  4 10:04:05 PDT 2018] Executing remove_dir shairport-sync
SSRPARI [Wed Apr  4 10:04:05 PDT 2018] Executing git clone https://github.com/mikebrady/shairport-sync.git
SSRPARI SSRPARI [Wed Apr  4 10:04:13 PDT 2018] Executing cd shairport-sync
[Wed Apr  4 10:04:13 PDT 2018] Executing autoreconf -i -f
SSRPARI [Wed Apr  4 10:04:49 PDT 2018] Executing ./configure --sysconfdir=/etc --with-stdout --with-pipe --with-alsa --with-avahi --with-ssl=openssl --with-metadata --with-soxr --with-systemd
SSRPARI [Wed Apr  4 10:05:21 PDT 2018] Executing make
SSRPARI [Wed Apr  4 10:05:47 PDT 2018] Executing make install
SSRPARI [Wed Apr  4 10:05:47 PDT 2018] Executing systemctl enable shairport-sync
SSRPARI [Wed Apr  4 10:05:47 PDT 2018] Running ./airplay_config.sh...
shairport-sync.conf
SSRPARI [Wed Apr  4 10:05:47 PDT 2018] Saving /etc/shairport-sync.conf...
shairport-sync.conf=/etc/
SSRPARI [Wed Apr  4 10:05:48 PDT 2018] Executing cp shScripts/shairportstart.sh /usr/local/bin/shairportstart.sh
SSRPARI [Wed Apr  4 10:05:48 PDT 2018] Executing cp shScripts/shairportend.sh /usr/local/bin/shairportend.sh
SSRPARI [Wed Apr  4 10:05:48 PDT 2018] Executing chmod +x /usr/local/bin/shairportstart.sh
SSRPARI [Wed Apr  4 10:05:48 PDT 2018] Executing chmod +x /usr/local/bin/shairportend.sh
SSRPARI SSRPARI [Wed Apr  4 10:05:48 PDT 2018] You should now reboot
pi@piguy:~/super-simple-raspberry-pi-audio-receiver-install $ 

```

Lauszus, 2018-04-07:
I got it working simply by updating my system:

```bash
sudo apt-get update && sudo apt-get upgrade
```

stephenblender, 2018-04-07:
I am very new to raspberry pi and debian, and did get it working. In addition to my prevuios experience I also did the install again on a freshly flashed OS, stretch. This time I did an apt-get update/upgrade before the super simple install. Same experience, failed first time like above and succeeded on an immediate re-run.

Further, the following day I decided to add airplay which I had left off on the first install. I ran the installer again and it failed, but succeeded on an immediate re-run. This time I did no save the error reports, sorry. I suspect I can recover them somewhere if someone is interested.

Thanks for all the hard work!

XtrmJosh, 2018-04-07:
Same result on first run, ran update and upgrade (which seemingly had no effect - all packages up to date), then re-ran the install script and it succeeded. Now working flawlessly.

nolandonley14, 2018-05-29:
Im also getting this error and have done the updating and upgrading as well as tried to run and re-run the install but still haven't got it working. Anyone have any other tips?

qsabin, 2018-05-29:
I'm having the same issue as nolandonley14, I ran update and upgrade and have re-run the program several times to no avail.

BaReinhard, 2018-06-30:
Hey Guys, thanks for the interest in the project. Unfortunately, due to time constraints I am unable to address current issues and debugging, maintenance on the shell scripts are hard and inefficient. As such I will be porting over these scripts to ansible playbooks. This will allow for easier choosing of what you want installed, less brittle code base, and better maintenance so that less people experience the errors.

Once I have feature parity I will be addressing documentation on installing with ansible. The goal here is to allow people to setup multiple devices at once by only issuing one command, additionally to allow multiple OS support with less debugging and maintenance, as well as allowing other developers more easily add functionality to this repo with simple yaml files as opposed to obfuscated shell scripts.

Any open issues I will keep open to allow other community members to help solve issues, but currently I don't have the time to do so. I apologize for the lack of support, but once feature parity with ansible is achieved I believe everyone will be happier and better off.
Thanks for understanding.

[image]

BaReinhard, 2018-06-30:
Unfortunately, the problem here may be due to how brittle the install scripts are (hence the change in installation candidates mentioned above). What you may have to do is start fresh and due an update and upgrade then install. For best results I would suggest using Jessie for the time being.

benjamingotke, 2019-08-26:
I still have this issue? Someone got any news??

emma-goldman, 2020-01-16:
> I still have this issue? Someone got any news??

if you're on later versions of raspbian then it might be because libjson0-dev is deprecated, try:
 
```
sudo apt-get install libjson-c-dev libjson-c3
sudo nano ./dependencies.sh
ctrl+w, find libjson0-dev replace with libjson-c-dev
ctrl+x, y, enter
sudo ./install.sh
```
I had some other problems encountered after I did this so not sure if it actually solved the issue, gonna try a yocto guide I found rather than troubleshooting this tho since I only really need the bluetooth A2DP sink. 

Or you could roll back your OS version to match the one this was written for. good luck c: