BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #123: What to use as audio device for pHat DAC?

davthomaspilot, 2018-03-29:
Will the script it work with this I2S DAC?

davthomaspilot, 2018-03-29:
Found this feature request:

[https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/115](url)

So, I think this means I just choose HifiBerry DAC as the option for the audio device?

davthomaspilot, 2018-04-02:
So, I can bluetooth connect from my Ipad to Pi  Zero W with PHAT DAC, but no audio plays.

Before I ran the install script here, I could hear sound using speaker-test and aplay.  But now I get messages about access denied when I try to do speaker-test and connection refused when I do an aplay.  

When I do pactl list sources  I see

 bluez_source.A4_F1_E8_B6_2E_72.a2dp_source

but it's in the suspended state.  Is that expected?

davthomaspilot, 2018-04-02:
sudo systemctl status bluetooth*
* bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-04-02 19:43:28 EDT; 1min 20s ago
     Docs: man:bluetoothd(8)
 Main PID: 356 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           `-356 /usr/lib/bluetooth/bluetoothd

Apr 02 19:43:28 ipad2gma340 systemd[1]: Started Bluetooth service.
Apr 02 19:43:28 ipad2gma340 bluetoothd[356]: Bluetooth management interface 1.14 initialized
Apr 02 19:43:28 ipad2gma340 bluetoothd[356]: Failed to obtain handles for "Service Changed" characteristic
Apr 02 19:43:28 ipad2gma340 bluetoothd[356]: Sap driver initialization failed.
Apr 02 19:43:28 ipad2gma340 bluetoothd[356]: sap-server: Operation not permitted (1)
Apr 02 19:43:28 ipad2gma340 bluetoothd[356]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource
Apr 02 19:43:28 ipad2gma340 bluetoothd[356]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSink
Apr 02 19:43:29 ipad2gma340 bluetoothd[356]: Endpoint registered: sender=:1.5 path=/A2DP/SBC/Source/1
Apr 02 19:43:29 ipad2gma340 bluetoothd[356]: Endpoint registered: sender=:1.5 path=/A2DP/SBC/Sink/1
Apr 02 19:43:29 ipad2gma340 bluetoothd[356]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)

davthomaspilot, 2018-04-03:
Didn't mean to close the issue.  Not sure how to open.  But, is anyone monitoring anyway?

lordfinal, 2018-07-14:
Hi, 

I want to use the phat dac too. But I just got them. Did you tried the method mentioned here? 

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/36

alexander480, 2019-01-20:
Run this command before installation
`curl https://get.pimoroni.com/phatdac | bash`

Reboot

Start installation

Select "No sound card" when promoted to select sound card