BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #125: No local audio from RPI

johans86, 2018-04-26:
Hi,
Playing audio on the RPI from my computer works well over Bluetooth, but I have not been able to play audio locally on the RPI after the installation. For instance there is no audio when starting a YouTube-video in the browser of Raspbian desktop.

I'm probably missing something simple.

**My setup**

Raspberry PI 3 model B
JustBoom Amp HAT
Raspbian Stretch with desktop

Best regards, Johan

gterras, 2018-04-30:
Hi, 

From what I've seen Pulseaudio is not launched automatically at boot, you can try to open a terminal and type `pulseaudio &`.

adrian-elian, 2018-09-28:
> Hi,
> Playing audio on the RPI from my computer works well over Bluetooth, but I have not been able to play audio locally on the RPI after the installation. For instance there is no audio when [...]

I've also encountered this. BT streaming works fine and i can control even the volume, but when i do 

`sudo aplay /usr/share/sounds/alsa/front_center.wav` no sound comes through the  speakers... 

@johans86 did you manage to find a solution ? 

For me `sudo aplay /usr/share/sounds/alsa/front_center.wav -D sysdefault ` plays music but at full volume and i can't control the volume in any way.  :(

johans86, 2018-09-28:
I didn't find any stable solution that worked every time from what I can
remember, and gave up on it. Would be great if it worked though!

-- 

Johan Seijsing
Lustgårdsgatan 5
112 18 Stockholm

CONFIDENTIALITY NOTICE: If you have received this message in error, please
notify the sender immediately by return email and delete this message from
your system. Thank you for your cooperation. This message and any attached
documents may contain privileged and confidential information. If you are
not the intended recipient, be aware that any re-disclosure, copying,
distribution or use of the information by any other party is strictly
prohibited.