BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #128: Pi compatibility

fibigod, 2018-05-16:
I have a Pi model B lying around. Has anyone used it with this model? Will it work?

BaReinhard, 2018-05-17:
It should work, only concern would be what version of raspbian you are using

fibigod, 2018-05-17:
I will try i this weekend with Jessie lite

fibigod, 2018-05-17:
install takes a long time. But it functions properly so far :-)

BaReinhard, 2018-06-30:
glad its working. Please reopen if you still have issues:

Hey Guys and Gals, thanks for the interest in the project. Unfortunately, due to time constraints I am unable to address current issues and debugging, maintenance on the shell scripts are hard and inefficient. As such I will be porting over these scripts to ansible playbooks. This will allow for easier choosing of what you want installed, less brittle code base, and better maintenance so that less people experience the errors.

Once I have feature parity I will be addressing documentation on installing with ansible. The goal here is to allow people to setup multiple devices at once by only issuing one command, additionally to allow multiple OS support with less debugging and maintenance, as well as allowing other developers more easily add functionality to this repo with simple yaml files as opposed to obfuscated shell scripts.

Any open issues I will keep open to allow other community members to help solve issues, but currently I don't have the time to do so. I apologize for the lack of support, but once feature parity with ansible is achieved I believe everyone will be happier and better off.
Thanks for understanding.

[image]

closed 2018-06-30 by BaReinhard