BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #129: Fatal error encountered

Marcookl, 2018-05-16:
Executing sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
Fatal error encountered: su volumio -c ./bt_pa_config.sh failed

capraDemon, 2018-05-18:
I'm experiencing the same issue. 
 SSRPARI [Fri May 18 05:06:41 UTC 2018] Executing sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
Fatal error encountered: su volumio -c ./bt_pa_config.sh failed

also: 
when executing this command by hand, it works
volumio@volumio01:~/super-simple-raspberry-pi-audio-receiver-install$ sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py

so I've commented out the regarding line, but the install script failed at the next copy command. 
I repeated the steps for that, but the installer seems to failed on every subsequent cp. 

Fatal error encountered: 'sudo cp init.d/pulseaudio /etc/init.d/pulseaudio' failed

I'm using a raspberry pi 3 b+ 
uname -a
Linux volumio01 4.14.29-v7+ #1101 SMP Thu Mar 22 17:27:30 GMT 2018 armv7l GNU/Linux

gewoondemax, 2018-05-28:
Hello all got the same issue - running Raspberry pi3 and Volume Image 29 

Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
Fatal error encountered: su volumio -c ./bt_pa_config.sh failed

s1m00, 2018-06-12:
As reported in #103 : 
- edit bt_pa_config.sh with "sudo nano" and change every "sudo" command in "sudo -S";
- delete "exc" tag before new "sudo - S chmod XXX" (where XXX are numbers of read-write permissions) and "sudo - S patch". Someone suggested to remove all, but in my case worked in this way
- launch again installation with "sudo ./install.sh"
- at a certain point (when first "sudo -S" command is met installing "volume-watcher.py") it will stop and seems to freeze. You have to type in your password for new "sudo -S" command and ENTER
I hope this will help you.

BaReinhard, 2018-06-30:
Hey Guys, thanks for the interest in the project. Unfortunately, due to time constraints I am unable to address current issues and debugging, maintenance on the shell scripts are hard and inefficient. As such I will be porting over these scripts to ansible playbooks. This will allow for easier choosing of what you want installed, less brittle code base, and better maintenance so that less people experience the errors.

Once I have feature parity I will be addressing documentation on installing with ansible. The goal here is to allow people to setup multiple devices at once by only issuing one command, additionally to allow multiple OS support with less debugging and maintenance, as well as allowing other developers more easily add functionality to this repo with simple yaml files as opposed to obfuscated shell scripts.

Any open issues I will keep open to allow other community members to help solve issues, but currently I don't have the time to do so. I apologize for the lack of support, but once feature parity with ansible is achieved I believe everyone will be happier and better off.
Thanks for understanding.

[image]