BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #130: No Airplay audio

wakeldakel, 2018-05-21:
Hi,
first of all thank you so much for this project! This is such a great idea.

I'm trying to install the Airplay and Bluetooth components onto a Pi Zero W with a DragonFly USB sound card attached. Bluetooth is working fine, but Airplay won't output any audio.
When I start streaming to shairsync CPU usage is at about 30%, so audio should be processed correctly. I already tried all other solutions from other issues, especially regarding the `/etc/asound.conf` file.

This is my `aplay -l` output:
```
**** List of PLAYBACK Hardware Devices ****
card 1: v15 [AudioQuest DragonFly Black v1.5], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

This is my current `/etc/asound.conf`:
```
pcm.pulse {
    type pulse
    card 1
}
ctl.pulse {
    type pulse
    card 1
}
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```

`/etc/shairport-sync.conf` isn't changed from the installed version.

Thanks in advance!

sose5000, 2018-05-23:
Was it simple to install on the Zero?  Are you planning to use it in gadget mode?  I'm looking to use a Pi to replicate the ipod protocol for streaming media and I am curious if anyone else has done it.

uzlonewolf, 2018-06-29:
I had the same issue.  Turns out /etc/shairport-sync.conf is incorrectly using "hw:0,0" - I changed it to "hw:1,0" and audio now works for both Bluetooth and Airplay.

BaReinhard, 2018-06-30:
Hey Guys, thanks for the interest in the project. Unfortunately, due to time constraints, maintenance on the shell scripts are hard and inefficient. As such I will be porting over these scripts to ansible playbooks. This will allow for easier choosing of what you want installed, less brittle code base, and better maintenance so that less people experience the errors.

Once I have feature parity I will be addressing documentation on installing with ansible. The goal here is to allow people to setup multiple devices at once by only issuing one command, additionally to allow multiple OS support with less debugging and maintenance, as well as allowing other developers more easily add functionality to this repo with simple yaml files as opposed to obfuscated shell scripts.