BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #133: Working fine on Volumio with android. Not discoverable with iPhone

wodgey, 2018-06-04:
Any ideas? My android phone connects and plays audio but my Mrs cannot see it from her iphone.

I have : -
Rebooted.
Disabled BT on Android, then rebooted again.
Bluetoothctl - set discoverable & pairable (even tried pairing to the iPhone from the raspberrypi. failed)
I noticed that iPhone says "legacy pairing - no". could this be it?

My dongle is BT v2. does this make a difference?

wodgey, 2018-06-06:
I've tested on other Android devices and they can all see the device whereas other users using ios cannot. I am baffled. I do not know what to google to further my understanding

BaReinhard, 2018-06-30:
Hey Guys, thanks for the interest in the project. Unfortunately, due to time constraints I am unable to address current issues and debugging, maintenance on the shell scripts are hard and inefficient. As such I will be porting over these scripts to ansible playbooks. This will allow for easier choosing of what you want installed, less brittle code base, and better maintenance so that less people experience the errors.

Once I have feature parity I will be addressing documentation on installing with ansible. The goal here is to allow people to setup multiple devices at once by only issuing one command, additionally to allow multiple OS support with less debugging and maintenance, as well as allowing other developers more easily add functionality to this repo with simple yaml files as opposed to obfuscated shell scripts.

Any open issues I will keep open to allow other community members to help solve issues, but currently I don't have the time to do so. I apologize for the lack of support, but once feature parity with ansible is achieved I believe everyone will be happier and better off.
Thanks for understanding.

[image]

mstr3336, 2019-01-27:
Ah tbh that's just because Apple's Bluetooth protocol is proprietary and requires devices to undergo some expensive signature thing before they are allowed to connect to iOS devices via Bluetooth.

Very unlikely to be a problem with the software, though, so much as an iOS "feature".
I understand that airPlay is implemented to handle iOS/ OSX devices, which might suit your needs.
@BaReinhard you can probably close this now.