BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #145: Volume Control

TypeDaniel, 2018-09-12:
Hi,

Volumio all up and running with the SSRPARI, but I can't seem to adjust the volume settings, its either muted or dialed to 11.

Is there a way I can control this?

xavier679, 2018-09-22:
The iPhone can't control the volume through bluetooth and doesn't understand what's going on.
Environment: volumio2.457, raspberry pie 3B+, bluetooth adapter

kilvn, 2019-02-23:
if can use,please  @ me.

ljj3029, 2019-10-07:
貌似和ALSA有关
我参考了https://wiki.archlinux.org/index.php/PulseAudio#Expose_PulseAudio_sources,_sinks_and_mixers_to_ALSA这个链接，但是我并不能实现，与我树莓派相连的显示器出现声音

ljj3029, 2019-10-07:
貌似和ALSA有关
我参考了https://wiki.archlinux.org/index.php/PulseAudio#Expose_PulseAudio_sources,_sinks_and_mixers_to_ALSA
这个链接，但是我并不能实现，与我树莓派相连的显示器出现声音

ljj3029, 2019-10-07:
貌似和ALSA有关
我参考了https://wiki.archlinux.org/index.php/PulseAudio#Expose_PulseAudio_sources,_sinks_and_mixers_to_ALSA这个链接，但是我并不能实现，与我树莓派相连的显示器出现声音

> 你好
> 
> Volumio已全部使用SSRPARI启动并运行，但是我似乎无法调整音量设置，将其静音或拨至11。
> 
> 有办法可以控制吗？

> Hi,
> 
> Volumio all up and running with the SSRPARI, but I can't seem to adjust the volume settings, its either muted or dialed to 11.
> 
> Is there a way I can control this?
sir，请问您解决了吗

xavier679, 2019-10-07:
很抱歉，这是很久前做的了，已经忘了

| |
解-xavier
|
|
xavier679@163.com
|
签名由网易邮箱大师定制