BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #146: Fatal error encountered

ReadyFoxy, 2018-09-29:
I get

> Fatal error encountered: 'sudo make' failed
> Fatal error encountered: su pi -c ./bt_pa_config.sh failed

After executing sudo ./install.sh

schwukas, 2019-02-19:
Do you still encounter this problem? I'd be glad to help.

ReadyFoxy, 2019-02-19:
I have not tested since

divx118, 2019-06-03:
@ReadyFoxy @schwukas I don't know if it is related, I use stretch on rpi3.
I got the following error.
```
SSRPARI [Mon  3 Jun 20:11:03 CEST 2019] Executing sudo ./bootstrap.sh
Fatal error encountered: 'sudo ./bootstrap.sh' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed
```
bootstrap.sh failed because autopoint was missing.
`sudo apt-get install autopoint`

Also got an earlier installing `libjson0-dev` failed, but that is replaced with `libjson-c-dev` on stretch. So easy fix.
After this fixing I got the:
```
SSRPARI [Mon  3 Jun 20:33:56 CEST 2019] Executing sudo ./bootstrap.sh
SSRPARI [Mon  3 Jun 20:39:30 CEST 2019] Executing sudo make
Fatal error encountered: 'sudo make' failed
Fatal error encountered: su pi -c ./bt_pa_config.sh failed
pi@raspberrypi:~/Super-Simple-Raspberry-Pi-Audio-Receiver-Install $ cd /home/pi/pulseaudio/
pi@raspberrypi:~/pulseaudio $ sudo make
make  all-recursive
make[1]: Entering directory '/home/pi/pulseaudio'
Making all in src
make[2]: Entering directory '/home/pi/pulseaudio/src'
make  all-am
make[3]: Entering directory '/home/pi/pulseaudio/src'
  CC       modules/raop/libraop_la-raop_client.lo
modules/raop/raop_client.c: In function ‘rsa_encrypt’:
modules/raop/raop_client.c:167:8: error: dereferencing pointer to incomplete type ‘RSA {aka struct rsa_st}’
     rsa->n = BN_bin2bn(modules, size, NULL);
        ^~
Makefile:8470: recipe for target 'modules/raop/libraop_la-raop_client.lo' failed
make[3]: *** [modules/raop/libraop_la-raop_client.lo] Error 1
make[3]: Leaving directory '/home/pi/pulseaudio/src'
Makefile:4806: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/pi/pulseaudio/src'
Makefile:799: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/pulseaudio'
Makefile:614: recipe for target 'all' failed
make: *** [all] Error 2
```
Just thought I should post this in case someone else comes across similar issues until https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/136 is finished.