BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #148: My feedback

shamitoH, 2018-10-24:
So here's my feedback. 
I'm using raspberry pi 3 with Raspbian Stretch.
Installed following the instructions above, so at initialization after first reboot get the message 

```
**[FAILED]** Failed to start LSB: Makes Bluetooth discoverable and connectable to 0000...
 See "systemctl status bluetooth-agent.service" for details.
```
getting the following:

```
pi@raspberrypi:~ $ systemctl status bluetooth-agent.service
● bluetooth-agent.service - LSB: Makes Bluetooth discoverable and connectable to 0000
   Loaded: loaded (/etc/init.d/bluetooth-agent; generated; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-10-24 01:27:28 UTC; 4min 4s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 409 ExecStart=/etc/init.d/bluetooth-agent start (code=exited, status=1/FAILURE)

Oct 24 01:27:26 raspberrypi systemd[1]: Starting LSB: Makes Bluetooth discoverable and connectable to 0000...
Oct 24 01:27:26 raspberrypi bluetooth-agent[409]: setting bluetooth discoverable
Oct 24 01:27:28 raspberrypi sudo[430]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/hciconfig hci0 piscan
Oct 24 01:27:28 raspberrypi sudo[430]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 24 01:27:28 raspberrypi bluetooth-agent[409]: Can't get device info: No such device
Oct 24 01:27:28 raspberrypi sudo[430]: pam_unix(sudo:session): session closed for user root
Oct 24 01:27:28 raspberrypi systemd[1]: bluetooth-agent.service: Control process exited, code=exited status=1
Oct 24 01:27:28 raspberrypi systemd[1]: Failed to start LSB: Makes Bluetooth discoverable and connectable to
Oct 24 01:27:28 raspberrypi systemd[1]: bluetooth-agent.service: Unit entered failed state.
Oct 24 01:27:28 raspberrypi systemd[1]: bluetooth-agent.service: Failed with result 'exit-code'.
```
The raspberry was visible in other devices but unable to connect. The issue can be solved by restarting the service, but you must do it every time the raspberry is restarted.

I realized that bluetooth-agent.service starts before Bluetooth service does. So edited the init.d file for bluetooth-agent. Adding some seconds (5 works fine for me) in the sleep line at the start header, there would be time to bluetooth service to start. So here's my /etc/init.d/bluetooth-agent:

```
#!/bin/sh -e
### BEGIN INIT INFO
# Provides:             bluetooth-agent
# Required-Start:       $remote_fs $syslog bluetooth pulseaudio
# Required-Stop:        $remote_fs $syslog
# Default-Start:        2 3 4 5
# Default-Stop:         0 1 6
# Short-Description:    Makes Bluetooth discoverable and connectable to 0000
# Description:          Start Bluetooth-Agent at boot time.
### END INIT INFO
#! /bin/sh
# /etc/init.d/bluetooth-agent
USER=root
HOME=/root
export USER HOME
case "$1" in
    start)
        echo "setting bluetooth discoverable"
        sleep 5
        sudo hciconfig hci0 piscan
        start-stop-daemon -S -x /usr/local/bin/simple-agent.autotrust &
        echo "bluetooth-agent started"
        ;;
    stop)
        echo "Stopping bluetooth-agent"
        start-stop-daemon -K -x /usr/local/bin/simple-agent.autotrust
        ;;
    *)
        echo "Usage: /etc/init.d/bluetooth-agent {start|stop}"
        exit 1
        ;;
esac
exit 0

```

I'm not an english speaker, so sorry if I use inappropriate language or inconsistencies.

prae, 2018-11-16:
Thanks- this was helpful!

old-square-eyes, 2019-03-17:
Manually restarting the service allowed me to connect once and play music. Then I changed the file `/etc/init.d/bluetooth-agent` as advised, rebooted and couldn't connect. No, manually resetting does nothing and I can't connect at all.