BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #150: Start pulseaudio with realtime and no CPU limit

tweinreich, 2019-04-04:
For me adding the `--realtime` and `--no-cpu-limit` to `init.d/pulseaudio` script as startup parameters solved some of my issues with lagging output (I may have to try a newer Pi and/or a decent sound card).

Tested with:
* Raspberry Pi 2011.12
* Raspbian Stretch
* No extra sound card / interface
* iPhone 5s (AirPlay)
* Motorola Moto Z (Default Bluetooth)

otmpp, 2019-06-25:
Have you done that to stop the chopping that happens when the onboard wifi is enabled while using WiFi at the same time?

tweinreich, 2019-06-25:
@otmpp If I'm correct Model A Revision 2.0 doesn't have onboard WiFi. I used an Edimax USB WiFi Adapter. It was activated the whole time though. Did you experience any issues with this?

otmpp, 2019-06-25:
Ahh yes I see what you mean. For bluetooth I have only ever used the Zero W or the 3B or 3B+ and I've used the onboard WiFi and Bluetooth. The only way which I have that I can stream bluetooth music reliably is by disabling the onboard WiFi since they interfere with one another. I have heard that you can use a usb WiFi card like you have and then like that it works. The way I have used his scripts is either on the Zero W or 3B. I have only tested the Bluetooth only installation and only once I disabled the onboard WiFi could i then stream audio well

tweinreich, 2019-08-13:
@otmpp ok that's not fortunate for me as I wanted to use WiFi because I hoped it would be more reliable and can cover a greater area. Thank you for sharing your experience.