BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #159: can't install on buster

sergio49, 2020-08-02:
Hi,
When installing on a rpi zero with buster, I got fatal error : 

Fatal error encountered: Installation of package 'libjson0-dev' failed

an d other libs as well...

How to pass through this please ?
Thanks

TysonBy, 2020-08-10:
Hello,
i have this problem too,
I testing this issues..

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/122#issuecomment-575206044

huncksas, 2020-10-21:
Hi,

i guess anybody who wants to analyse this needs some more informations about what you did before the issue occoured.

Me for myself ran through the single steps only for bluetooth setup ( so i can pair a device and stream music ).
There, i have had this issue also - just removed the "libjson0-dev" - package.  Another warning poped up that another package will be replaced with another one.

But at the end, removing "libjson0-dev" in the dependencies-file worked fine for me and did what i expected.

Zorro122, 2021-03-13:
Faced the same issue on buster, replacing libjson0-dev by libjson-c-dev fixed the first issue, but then I faced another fatal error.

`fatal error patch /etc/bluetooth/main.conf`

In the end, I've modified the install.sh file by commenting (using "#") both lines in which bt_pa_config.sh was mentioned.
Worked fine in the end. Bluetooth name is not changed to the enterd name, but at least I can play music now from my phone on the Raspberry.